wled/WLED #653: HTTP request XML

GaseGasovic, 2020-02-02:
Hi guys. Im a newbie at WLED comunity and programming. I just want to ask you if there is a way to code a simple html page with several buttons to act as a some sort of "user interface" with just few pre-requested effects. I tried to do so, but I get a XML values as a response. Is there a way to disable a XML response? Thanks in advance. :)

Aircoookie, 2020-02-02:
Hi and welcome!
If you do not need a custom user interface, presets are probably the right way to go (numbers 1-16 in favorites tab). If you do want custom HTML, just call `[esp-ip]/win&PL=1` to load the first preset for example :)

Def3nder, 2020-02-03:
Hi @GaseGasovic, 
if I understood right, your problem is that sending something to the HTML-API does result in an "answer" in return that will be displayed by the browser - is this right ?

And: no: currently this cannot be "suppressed".

@Aircoookie, what about an addition API switch that - if set - will not send anything in return ?

GaseGasovic, 2020-02-03:
Hi @Def3nder, 
Yeah, thats right. The problem is answer shown by browser. So for now, as I can understand, there is no way, to turn off a "answer" sending feature? @Aircoookie that idea about an API switch from @Def3nder side is really great. Maybe in later updates it will be possible. :)

Aircoookie, 2020-02-03:
No, that's not the way APIs work in general. You are not supposed to interface with them directly via user navigation, but instead in the background. I suppose you are trying to just open links that are the API request. Even if I "turned the response off", the browser would just load an empty page.

Try putting this in your html <head> instead:
```js
<script>
function SendReq(strA)
{
  nocache = "&nocache=" + Math.random() * 1000000;
  var request = new XMLHttpRequest();
  // send HTTP GET request
  request.open("GET", "win" + strA + nocache, true);
  request.send(null);
}
</script>
```

and in your HTML something like this, you just call the javascript function:
```html
<button type="button" onclick="SendReq('&PL=1')">Preset 1!</button>
```

Aircoookie, 2020-02-03:
If you are not hosting the custom page on the esp, you'll need
```js
request.open("GET", "http://[ESP IP address]/win" + strA + nocache, true);
```
instead

GaseGasovic, 2020-02-03:
@Aircoookie and @Def3nder thank you guys so much. It helped to solve my problem. Thanks for an awesome software @Aircoookie and Im waiting to see whats comming in future. Cheers, lads. :)

closed 2020-02-03 by GaseGasovic