wled/WLED #655: stairway wipe effect last 10% of leds not playing well

haimbilia, 2020-02-03:
i installed the stairway wipe effect usermod, it works great but the about 10-15% at the end of the strip will just turn on all together on wipe-on (U0=1).
and on U0=0 they will act the same, turned off all together (not one by one).
i have managed to make it work better by increasing the number of led pixels in the settings (about 15% more).
but that messes with my other effects.
don't know how it can be resolved but will be glad to test if anyoune has any idea.

Aircoookie, 2020-02-03:
Hi, thank you for the report. What is your LED count? I believe the usermod might be doing some rounding it shouldn't do...

haimbilia, 2020-02-03:
i tried it with a 100 leds strip and a 15 leds strips. the results are similar.

huggy-d1, 2020-02-11:
Try telling a lie to WLED and add 10% more LEDs than you really have.  What happens then?

huggy-d1, 2020-02-11:
Also, when you build the .bin firmware, was this commented out:
#define STAIRCASE_WIPE_OFF
I wonder if modifying the If/else if's to a switch case structure would help things out (depending on the compiler used perhaps):
```
switch (wipeState)
{
	case 0: 
		startWipe();
		wipeState = 1;
		break;
	case 1: //wiping
		uint32_t cycleTime = 360 + (255 - effectSpeed)*75; //this is how long one wipe takes (minus 25 ms to make sure we switch in time)
		if (millis() + strip.timebase > (cycleTime - 25)) { //wipe complete
			effectCurrent = FX_MODE_STATIC;
			timeStaticStart = millis();
			colorUpdated(3);
			wipeState = 2;
		}
	case 2: // static
		if (userVar1 > 0) { //if U1 is not set, the light will stay on until second PIR or external command is triggered
			if (millis() - timeStaticStart > userVar1*1000) wipeState = 3;
		}
	case 3: //switch to wipe off
		#ifdef STAIRCASE_WIPE_OFF
			effectCurrent = FX_MODE_COLOR_WIPE;
			strip.timebase = 360 + (255 - effectSpeed)*75 - millis(); //make sure wipe starts fully lit
			colorUpdated(3);
			wipeState = 4;
		#else
			turnOff();
		#endif
	default: //wiping off
		if (millis() + strip.timebase > (725 + (255 - effectSpeed)*150)) turnOff(); //wipe complete
}
```

closed 2020-06-17 by stale[bot]