wled/WLED #656: SYNC Delay - Feature Request

atuline, 2020-02-03:
To add a variable delay in a SYNC'ed series of displays, each 100ms (or whatever) longer than the other.

You then change the first display in that subnet/UDP group and have it SYNC and they all change in a co-ordinated sequence . . . just like show dancers.

I would also add a flag, so that they either SYNC at the same time, or with that pre-assigned delay.

yusufsulaiman, 2020-02-05:
hi @atuline , i have try to 4 IP address Sync - WLED Preset Cycle like this  video
https://www.youtube.com/watch?v=jU0caygPwSA

atuline, 2020-02-05:
Hi @yusufsulaiman - Wow! Nice big panels. I'm on a poverty budget. Here's a version I made with my own MQTT code and FastLED:

https://www.youtube.com/watch?v=jX91v_ByoaY

If I were to continue that, I'd come up with several delay waveforms.

Def3nder, 2020-02-05:
Hi @atuline, that's not the way UDP sync does work.

It's a "one to many" connection with one sender and numerous receivers.
What propably could work is to add a specified amount of time on every single WLED instance - then each instance would delay the sync by the configured amount of time.

I've just tested this:
In `wled07_notify.ino`, lines 167++:
```
    //wled notifier, block if realtime packets active
    if (udpIn[0] == 0 && !realtimeActive && receiveNotifications)
    {
      //ignore notification if received within a second after sending a notification ourselves
      if (millis() - notificationSentTime < 1000) return;
      if (udpIn[1] > 199) return; //do not receive custom versions
      delay(1000);
```
(I added the `delay(1000)`) - this has the desired effect that this WLED instance does delay by 1 second, if it receives an UDP notification.

BUT: this also results in a "1 second frozen effect" on that WLED instance, since it does not do any updates to the LED strip in that time.

So: in general this does work, but it would need more to integrate this seamless.

atuline, 2020-02-05:
Thanks for that effort. That's essentially how I did it with my own MQTT/FasLED project (and I really hate delay statements). Each device had a pre-defined delay value after it had received the signal. Now that you've shown me where the code for this is in WLED, I'll play around with this myself some more.

strfngr, 2020-05-24:
Hi @atuline, were you able to solve this?
I'm about to install ~30m of SK6812 (1800 leds) - So I guess I'll have to use multiple ESP8266s, but I'd like it to look like a seemless sequence

atuline, 2020-05-24:
> Hi @atuline, were you able to solve this?
> I'm about to install ~30m of SK6812 (1800 leds) - So I guess I'll have to use multiple ESP8266s, but I'd like it to look like a seemless sequence

Umm, I ended up going off this track and started to work on audio reactivity. We now have an audio reactive fork of WLED. Hopefully, I'll get back to this one of these days, but I expect it'll be via an external app/interface and will send these commands via the HTTP API. Oh, and @strfngr, I'd probably be looking to use ESP32's if possible. After all is said and done with WLED, there's NOT a lot of overhead left with ESP8266's, and I say that as someone with 40 of them. .

Legsmaniac, 2020-05-24:
> and I say that as someone with 40 of them. .

That means you've got 40 of the wrong/old ones now then. 🙂

atuline, 2020-05-24:
> > and I say that as someone with 40 of them. .
> 
> That means you've got 40 of the wrong/old ones now then. 🙂

Yea, no doubt I do, but the price was awesome on my poverty payment program of a budget.

Legsmaniac, 2020-05-24:
Lol. Indeed, but I was trying to point out that there are better 8266s out there now giving the much needed overhead so we don't have to be limited to 32s.  The Wemos D1 Mini Pro 16M for example. 🙂

atuline, 2020-05-24:
> Lol. Indeed, but I was trying to point out that there are better 8266s out there now giving the much needed overhead so we don't have to be limited to 32s. The Wemos D1 Mini Pro 16M for example. 🙂

Interestingly, that 2nd core of the ESP32 (not ESP32S) is proving to be critical, now that we've added the arduinoFFT library to the mix.

Legsmaniac, 2020-05-24:
The very best I've used is the Teensy 4. The speed! Wow! Sadly, no WiFi.

closed 2020-09-30 by stale[bot]