wled/WLED #657: One or two random LEDs come on after stretches of inactivity

CaptainUns, 2020-02-04:
I've been noticing that if WLED is switched off for more than 24-48 hours, but power is still being supplied to the controller (I'm using a D1 Mini), that a random LED or two near the beginning of the run (typically within the first 5 LEDs) will come on in either green or red. Switching WLED on and off works to turn these random LEDs back off, so it doesn't seem to be that the LEDs themselves are faulty, also because the next time it happens it might be a different LED or different pair of LEDs, so it appears to be some kind of software issue or bug. This particular run is permanently attached to my roof line, so its pretty inconvenient for random LEDs to come on when they are all supposed to be off. Anyone else have an issue like this? Hoping maybe this is a known bug that's been addressed and I just missed it in another post. Thanks!

recliq, 2020-02-04:
How long are the cables between the D1mini and the strip? Sounds more like a interference problem where electromagnetic waves induce a current in the wire which gets interpreted by the leds as signal to turn on.
WLED doens't send constant 'turn LEDs off' commands...

CaptainUns, 2020-02-04:
The distance to the first LED is about 10 feet, but the signal runs through a logic level shifter directly out of the D1mini first. The wire isn't particularly heavily shielded though so I'll see about swapping it out for something a little more substantial.

sansillusion, 2020-02-04:
Also you could use a relay to cut power to the LED's when WLED is turned
off. I think there is already a usermod for this.

Def3nder, 2020-02-05:
Hi @CaptainUns, the is a know issue with WS28xx type LED strips (SK68xx as well).

The strip has power all the time and only the LEDs are all set to brightness of "0".
As the strip is not shielded in any way, it can catch up radiation that leads to a random signal on any of the LEDs. As each LED does hold it's state until it is changed by the shift register, this random color stays there until you activly clear the LED strip.

The best way is using the **replay PIN** to drive a relay that siwtches the power line off.

Have a look into `NpbWrapper.h, line 10`:
`#define RLYPIN 12 //pin for relay, will be set HIGH if LEDs are on (-1 to disable). Also usable for standby leds, triggers,...`

So all you need is **a little relay or a MOSFET** that can be driven from the ESP PIN.
Connect it to `PIN 12 `and you're done.

closed 2020-06-11 by stale[bot]