wled/WLED #661: JSON API command crashes ESP

Aircoookie, 2020-02-05:
The (illegal) JSON API call

```
{"seg":{"1":{"fx":2}}}
```

seems to crash the ESP (reported by @DutchmanNL on discord)

Def3nder, 2020-02-05:
Hi Aircoookie, 
sending the (wrong) above string **does not crash an ESP32 nor does it crash an ESP8266**.
I tried both: sending it to `/json/state` and to `/json`.

And sending 
```
{
    "bri": 50,
    "seg": [
        {
            "id": 0,
            "fx": 2
        }
    ]
}
```
to `/json/state` does **work as expected** on both ESP32 and ESP8266.

DutchmanNL, 2020-02-05:
Hmm interesting, did you try just ones or multiple times after each other ?

I saw this behavior by building an integration by JSON api and my system crashed the esp definitely  but not instand at the first one.

Using latest version build from git

Aircoookie, 2020-02-07:
Indeed I cannot reproduce the crash either, even with sending the request multiple times in a row.
@DutchmanNL maybe it was related to the communication rate (although I don't believe you send more than 10 requests a second, which is about the max. an ESP8266 running WLED can handle)
or some headers being sent, not the request itself.

closed 2020-02-07 by Aircoookie

DutchmanNL, 2020-02-07:
I can't reproduce it either only by indeed spamming the devices.
I keep it o bad coding on my side but can't verify as already improved :/