wled/WLED #663: Settings lost after power off

muxa, 2020-02-05:
For some reason when I turn off power supply and then turn it back on all the settings are lost (i.e the board is not connecting to WiFi and create a AP instead). 

Using Wemos D1 Mini, with a 12V WS2811 strip, D1 is powered with 5V using a 12 to 5V step down. 

However my second setup with a 5V strip and power supply is working fine, remembering all settings after restart. 

Any suggestions?

raoulteeuwen, 2020-02-05:
I also have varying success with saving settings ... did you try to change a setting in Config, hit save, go to 'Security & Updates' in Config and hit 'Save & Reboot'.

Aircoookie, 2020-02-06:
Did you use the same binary for both controllers? It might not work if you have accidentally flashed a 1M binary instead of the standard one to the D1 mini.

Another suggestion that might be able to fix it would be erasing the whole flash with
```
esptool.py erase_flash
```

huggy-d1, 2020-02-07:
> For some reason when I turn off power supply and then turn it back on all the settings are lost (i.e the board is not connecting to WiFi and create a AP instead).
> Using v0.8.6 on Wemos D1 Mini, with a 12V WS2811 strip, D1 is powered with 5V using a 12 to 5V step down.
> ...  suggestions?
This suggestion is probably wrong and not even possible with the hardware/software being used.  However, there is value in assuming there is a physical build problem, and then visually inspecting to prove there is none.

With bright light and magnifying glass, compare the 2 D1 Mini's.  See if there is a blob of solder that once it gets warm might issue a reset signal to WLED that takes effect on power up.  See if any wires have a loose strand sometimes going astray and possibly sending a signal that tells WLED to reset itself (in case WLED wifi credentials change, there has to be a way to force it into AP mode!!).

I think you will find the problem if you know it simply must be there.  I wish I knew what pin(s) WLED is reset by.  Or, maybe the bad D1 Mini has a bad reset switch.

Def3nder, 2020-02-09:
Hi @muxa, writing to the EEPROM does only work if the binary was built for exactly that flash size.
I had the same issue with an ESP8285 (with 1MB (or 8 Mbit) flash) as I used a WeMos D1 mini binary.
It did flash and WLED did run "normally" - the only effect was that writing to the EEPROM did not work. 
There are several versions of D1 mini's flouting around. I have some with 4MB flash and others with 16MB flash /the pro version).

You could have a look what esptool.py is reporting about that ESP:
`esptool.py --port COMx flash_id` 

Do you compile the binary yourself ? Or do you use the ones from the releases ?

@Aircoookie: we could add a check in the init function that checks, if the flash size used for the binary is the fitting to thenhardware:

```
    String realSize = String(ESP.getFlashChipRealSize());
    String ideSize = String(ESP.getFlashChipSize());
    bool flashCorrectlyConfigured = realSize.equals(ideSize);
```

spiff72, 2020-02-11:
I am very new to this piece of software, but I just ran into this today, too (I think).  I am using the latest released binary (0.8.6), and I grabbed the "WLED_0.8.6_ESP8266.bin" file to flash.  I am using a NodeMCU ESP8266 (from this link):
https://www.amazon.com/ESP8266-NodeMcu-Internet-Development-Micropython/dp/B07L8W9SP3

I have been working on over the last couple of days (designed a printed enclosure and soldered it all up).  I am powering it with a 12V power supply that feeds my 12v string of LEDs, and I have a MP1584EN-based DC-DC buck converter that drops 12v to 5v to power the Vin pin of the NodeMCU (and a level shifter for the data feed to the LEDs).

It was working fine this afternoon (unplugged it several times) and the presets would load as expected, and wifi connected fine.  I then unplugged it, left for an hour or two, and when I plugged it back in the app wouldn't connect (it was in AP mode).

I am building another one of these soon, so I want to know if there is anything i should be looking for.

spiff72, 2020-02-11:
So to clarify my issue from the original posts issue, mine is intermittent, or perhaps occurs after lengthy off time.

muxa, 2020-03-26:
I have reflashed with the latest 0.9.1 and it fixed it. Not sure if it was a process of reflashing or a new version that fixed it.

closed 2020-03-26 by muxa

tomh05, 2023-01-18:
I got this problem but it was because I'd flashed the 4MB image to a 1MB device. https://github.com/Aircoookie/WLED/issues/2612 Flashing the correct image fixed it.

krisrok, 2023-10-20:
@spiff72 
> It was working fine this afternoon (unplugged it several times) and the presets would load as expected, and wifi connected fine. I then unplugged it, left for an hour or two, and when I plugged it back in the app wouldn't connect (it was in AP mode).

I‘m seeing the same issue with a D1 mini. Were you able to sort it out?

spiff72, 2023-10-20:
> @spiff72
>
>
> I‘m seeing the same issue with a D1 mini. Were you able to sort it out?

It's been a long time since then, but I have moved on to ESP32 boards for WLED projects, so I don't think I am much help.

krisrok, 2023-10-20:
> It's been a long time since then

Certainly! Thanks for replying anyway.