wled/WLED #667: multiple WIFI configs

pille, 2020-02-08:
it would be useful to have multiple WIFI configs present.
the device may try to connect to them one-by-one until it succeeds or timeouts.
the benefit would be that you can have your lights at home and simply use them for something different, occasionally. e.g. a party/performance at a different location (a.k.a. different WIFI).

another use-case would be a test-network, which will be preferred, that is only activated when you want to work on your lights.

sjude68, 2020-02-12:
yes this would be a good feature to add.  I think Wled drops down to AP mode if configured WiFi is not available.  It would be nice to try an alternate wifi before AP mode.

Can one Wled module act as an AP and other modules can connected to it as clients.  How many clients can one Wled AP connect to.

Aircoookie, 2020-02-13:
> Can one Wled module act as an AP and other modules can connected to it as clients. How many clients can one Wled AP connect to.

Yes this is possible, however there might be a performance/stability impact. An ESP8266 can have 4 simultaneously connected clients, so 3 other modules if you also want an end device connected for control. Keep in mind though that you cannot directly access the client devices, they can only sync via UDP to the module that hosts the AP. What I'd recommend though is having some cheap portable router that opens a network, the AP is really only designed for setup and on-the-go control of that one light.

sjude68, 2020-02-13:
Suppose I want 2 or 3 Wled modules and 1 RaspberryPi using LEDFX for sound reactive control.  Is it possible to use 1 Wled in AP and the other in client mode.  RPi also in client mode.

Aircoookie, 2020-02-13:
@sjude68 that might be possible in E1.31 multicast mode, but I would recommend using the RPi as the AP instead as it has much more horsepower and there will be quite a bit of network routing required for E.13 live data.

sjude68, 2020-02-13:
Yes i think Rpi as AP is a better solution. Thanks.

Aircoookie, 2020-04-17:
There are so many issues now I don't notice some feature requests had already been made anymore. Closing this, tracking in issue #852 now :)

closed 2020-04-17 by Aircoookie