wled/WLED #673: Low FPS when changing state

pedrovwfreitas, 2020-02-11:
The effect is working correctly but the fade in/out between colors takes 2 seconds at 1 FPS

Def3nder, 2020-02-11:
Hi @pedrovwfreitas, could you please describe what exactly you did (which effect was running and which state did you change) and what you noticed ?

Which ESP board are you using and which version of WLED did you install ?

Did you use a pre-compiled release or did you compile yourself ?

pedrovwfreitas, 2020-02-12:
ESP8266(QuinLed Dig Uno), version 0.8.5 pre-compiled

https://www.youtube.com/watch?v=kRaSnzct4N4

Def3nder, 2020-02-12:
Hi @pedrovwfreitas, thx for sharing the video - this is really good to see the effect.

Hmm - in fact this a very slow transition. Is there a reason you use 0.8.5 ?
Could you please try the same with [0.9.0](https://github.com/Aircoookie/WLED/releases/tag/v0.9.0-b1) ? 

Which version of Home Assistant do you use?
...since 0.102 there is a native WLED integration for HA.

pedrovwfreitas, 2020-02-12:
I am now using 0.9.0 and it's working correctly but just the first led turns on

pedrovwfreitas, 2020-02-12:
Now everything is working correctly but I had to update by usb because thru OTA just the first led was turning on

closed 2020-06-11 by pedrovwfreitas