wled/WLED #674: WLED on a motorcycle! Ideas?

Andriejus, 2020-02-12:
[image]

Hello! I want to add 2 or 3 segments of LEDS to my motorcycle! Just thinking what features should I add?  So my questions are:

1. Will it be possible to control LEDs without wifi? Thinking to use phone app or just switches or rotary encoder to control everything. 
2.  1. 2, or 3 segments? With segments I could achieve cool effects. But not sure if it will be easy to control with buttons and maybe its not worth to divide into segments?

There are more questions, but for beginning I would like to hear some advices to begin with.  Mostly Im thinking how to control everything(buttons or rotary encoder) on a motorcycle and maybe you know some special features I could add??:)  Thanks!

huggy-d1, 2020-02-12:
My comment relates to it being legal in areas you will ride your motorcycle with the LEDs performing effects.  In some areas, many effects available with WLED are illegal.
That said, I think you could configure and use a button input to toggle through presets which have been preconfigured to those acceptable for use on a motorcycle on the roads, or perhaps a switch input that enables / disables effects depending on whether it is parked or riding down the road.

raoulteeuwen, 2020-02-12:
>     1. Will it be possible to control LEDs without wifi?
I wouldn't be surprised someone else replying a rotary dial is possible, just checking: you know that WLED will initially set up a local AP? So no need for any other network to be around: you can connect your phone (using wifi) to the 8266/ESP32 on which you run WLED and control WLED either via a browser or the WLED app.

aliakseuby, 2020-02-12:
https://youtu.be/g-Yzp2bwhyE
Делал на кольцах, в планах прицепить поворотники

Andriejus, 2020-02-13:
> https://youtu.be/g-Yzp2bwhyE
> Делал на кольцах, в планах прицепить поворотники

I will write in English, but I can speak russian:) Did you used these rings like in photo? I gave 2 of those and one bigger. Just wondering if its hard to install into lights?:)
Did you made with button to control effects?
[image]

aliakseuby, 2020-02-13:
> > https://youtu.be/g-Yzp2bwhyE 
> > Делал на кольцах, в планах прицепить поворотники
> 
> Я буду писать на английском, но я могу говорить по-русски :) Вы использовали эти кольца, как на фотографии? У меня есть 2 из них и один больше. Просто интересно, сложно ли его установить на свет? :)
> Ты сделал с помощью кнопки для управления эффектами?
Кольца как у тебя.
Немного добавил тюнинга (купил прозрачную эпоксидную смолу+люминофор) вырезал формы по диаметру колец, после выключения остается эффект свечения ) + это служит как рассеиватель и выглядит намного круче чем просто светодиоды ( как вариант можно сделать накладки из белого оргстекла, вырезать можно на лазерной резке чпу) 
Включение подцепил просто на габарит, включаются мгновенно, в настройках сделал запуск из сохраненного режима, если хочешь другой подключился по вай-фай настроил сохранил..
В планах сделать еще от кнопок (повороты) но не могу понять как сделать (после выключение поворота) в исходный режим
Фара разобралась довольно легко (фен) греть по соединению и аккуратно разъединить 
Клеил глазки на прозрачный герметик 
[image]
[image]

Quato38, 2020-02-14:
This sound like an interesting project but my russian is not so good.. can you translate det last part :-)

jezmck, 2020-02-27:
Google translate:

Rings like you.
 Tuning added a little (I bought a transparent epoxy resin + phosphor) cut out the shapes according to the diameter of the rings, after the shutdown there is a glow effect) + this serves as a diffuser and looks much cooler than just LEDs (as an option, you can make overlays from white plexiglass, you can cut it by laser cutting  cnc)
 The inclusion was picked up simply by the envelope, it turns on instantly, in the settings I made a start from the saved mode, if you want another connected via wi-fi configured saved.
 The plans do more from the buttons (turns), but I can’t understand how to do (after turning off the turn) to the initial mode
 The headlight was sorted out quite easily (hair dryer) to warm the connection and gently disconnect
 Glued eyes on a transparent sealant

closed 2020-07-04 by stale[bot]

Latifuni, 2020-11-22:
I have a Harley and I would not change anything there.

Jaketka, 2020-11-29:
Its cool