wled/WLED #675: Feature request: Option to broadcast current preset between WLED instances

ruddy17, 2020-02-13:
I'd like an option to enable sending/receiving broadcast of **preset slot load** event (in addition to brightness, color and effects we have now). This way one could set some matching (but not same) patterns and colors and switch them with one action.

Aircoookie, 2020-02-13:
Thank you for the suggestion!
This is indeed an interesting idea, but we would have to make it optional so that by default all lights still sync to the same setting. Then, this would be a great way to setup matching scenes.

blazoncek, 2022-02-21:
UDP syncing has now been enhanced with complete state syncing (including segment options and/or set up).
Unfortunately preset syncing may not be easily implemented but current sync (in yet to be released 0.13-b7) fulfils basic request as long as target devices have the same WLED version or newer.
Closing.

closed 2022-02-21 by blazoncek