wled/WLED #676: Issues with connection and json

profi2111, 2020-02-13:
Hey :)

I just discovered WLED and wanted to adopt it to my SK6812RGBW LED system. I set it up on a NodeMCU Lua Lolin V3 Module ESP8266 ESP-12E.
 

([https://www.amazon.de/AZDelivery-NodeMCU-ESP8266-ESP-12E-Development/dp/B074Q2WM1Y/ref=sxin_2_ac_d_rm?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&ac_md=0-0-bm9kZW1jdQ%3D%3D-ac_d_rm&cv_ct_cx=nodemcu&keywords=nodemcu&pd_rd_i=B074Q2WM1Y&pd_rd_r=9b29e026-0892-4d1a-84db-3ccc430f4d76&pd_rd_w=mAaVM&pd_rd_wg=ivoyr&pf_rd_p=27165605-abba-4b2d-a048-7f1b68c397ce&pf_rd_r=SCVJE29KN7HZE49WADNB&psc=1&qid=1581593226&sr=1-1-fe323411-17bb-433b-b2f8-c44f2e1370d4](url) 

Now... I uploaded the binary file via ESP HomeFlasher 1.2.0.
So far so good, the ESP booted and i added it to my wireless network. Now i had a look at some settings, played a bit with the colourwheel and the effects without a LED strip connected.
Now i had two issues. 

1. Error message "Type error: Network Error when attempting to fetch resource"
2. Error message "Type error: json is null"

I connected the ESP to my LED strip for testing if i get an output, but nothing. I use the SK6812RGBW strips and activated the option with the white LED.

Also tried compiling via Arduino and AP-Mode, same result and error messages. 

[image]
 #[image]

Am i using the wrong hardware?
Would be nice if i get a fast response :)

Yours sincerely
Christian

srg74, 2020-02-13:
Hi @profi2111 
Looks like board is ok. Have to be similar to NodeMCU variant. You did not mention about what firmware you used e.g. precompiled or you compile on our own?

Aircoookie, 2020-02-14:
Hi!
This was a bug in 0.9.0-b1 I have since fixed (gosh I really have to release a new version haha)
In the meantime, try this binary, which shouldn't have the issue anymore :)
https://drive.google.com/open?id=1GeWCOK1BfCeUIzwl-zL3zzf-gHnhg4zp

raoulteeuwen, 2020-02-14:
@profi2111 if after trying the latest release Aircookie wipped up for you doesn't light your LED's, can you let us know and also describe how you've exactly your components and hooked them up, so PSU, which wires run to what pin etc on what component etc. If you can attach a picture or diagram, that could help. And share what kind of experience or knowledge you have with hooking up stuff like this. Thank you.

profi2111, 2020-02-14:
@Aircoookie 
Thanks so much for you help, i'll try it later and write if it worked. Do you also have the new binary for an esp32? I am about to build the QuinLED board :)

Thanks again you are the best 👍 

Christian

heavydetail, 2020-02-15:
@Aircookie thanks for your work, but please release a working new version or tell us what we have to change to get rid of this error (any commit?). If I pull the newest version its not building (same hardware esp8266 12-e, same LEDs): 

`wled18_server.ino:120:4: error: no matching function for call to AsyncCallbackJsonWebHandler::AsyncCallbackJsonWebHandler(const char [6], initServer()::__lambda15)'});`

closed 2020-06-23 by stale[bot]