wled/WLED #677: Can I move my WLED's to a seperate VLAN?

fribse, 2020-02-13:
Hi Aircookie

I've started the motion of my NOT's and IOT's to seperate VLAN's.
It works well for most items, but I'm a bit unsure about WLED, and before I just cut off the branch I'm sitting on, I wanted to check in with you :-)
I have it integrated to HA, but as I understand the integration, the HA connects to WLED's API, and therefore it doesn't need any ports open from the NOT's VLAN to the LAN?
But how about the WLED discovery, both from the app and the HA, is that just done with mDNS, or how is that accomplished? If it's 'just' mDNS, then I can just move them, because that is already working for my esphome devices.

fribse, 2020-02-14:
Ok, I tried doing this morning, so far everything is good, I removed them from HA, and put them back in with the new IP's. I couldn't autodiscover them though, but the app finds them without problems, so I guess it's working 😄

Aircoookie, 2020-02-14:
Glad it's working! Yep, mDNS and normal HTTP GET/POST is everything WLED needs to function, so it should work in a VLAN if devices in it are accessible from your personal device VLAN.

closed 2020-02-17 by fribse