wled/WLED #679: Nightly Builds?

mike811, 2020-02-14:
Would be great if some sort of nightly build binaries could be available...

SerhiiZhilin, 2020-02-17:
Yeap, would be great. On last pre-release there is no option to chose IR type. Just wired all up but no that option and can not test.

pille, 2020-02-18:
travis is already building different targets, so it may publish them.

santa81, 2020-04-10:
+1 
push, is there any link to nightly builds?

salopette, 2020-04-10:
Würde mich auch freuen, wenn es die nightly builds als .bin geben würde!!!

pille, 2020-04-20:
https://docs.travis-ci.com/user/deployment/releases/
needs to be implemented by @Aircoookie, because of auth tokens

Fonta, 2023-01-12:
Guess https://github.com/wled-install/wled-install.github.io can be used for inspiration. Seems fairly simply.

grischard, 2023-03-07:
The github action test builds produce artifacts, which can be downloaded and flashed. You want the PlatformIO CI action, and most probably only the runs on the `main` branch:

https://github.com/Aircoookie/WLED/actions/workflows/wled-ci.yml?query=branch%3Amain

The 8 firmwares built by one particular github action run are at the bottom, like so: https://github.com/Aircoookie/WLED/actions/runs/4338132057#artifacts

blazoncek, 2023-03-07:
It would be great if someone would make changes to the installers to use those.
PRs are welcome.

closed 2025-07-08 by DedeHai