wled/WLED #684: WLED & SK9822 -> Flickering

karimidze, 2020-02-15:
Hi!

I've put on wemos lolin nodemcu v3 with my SK9822 LED strip and WLED 0.9.0-b1 premade .bin
Pins used: d3, d4 (CLK and DATA respectively), no level shifter are used.

At some effects appears flickering, but not at all of them.
I've managed to forward hyperion packets and situation are repeated. Video in attach.

Can anybody help me?

https://drive.google.com/file/d/1yx4OikD9ZZcpX-FcOMWp1od2_WiBLSZB/view?usp=sharing

mike2nl, 2020-02-17:
@karimidze 
Have seen your video but could not see it really good.
Try to it in some steps and possible it helps you:

1. Better power supply because the one i used was damaged in one way or the other. Check it.
2. The setting for the max. current in config->led settings were not optimized
3. Started calculating all power consumtion because the current setting is more an advise as the real needs. That's OK to get the idea. That is NO mistake from @Aircoookie.
4. The data connection from the nodemcu can be the issue (perfect solderd, not a to long cable?)
5. The nodemcu is a real LOLIN? No Clone? On clones you have a lot of issues. Cheap hardware much cheaper. A VW Polo is no Ferrari ;-)
6. Try to add a 270, 330 or a 470 ohm resistor between the nodemcu and the data input on the led strip.  The SK9822 can be a bit tricky.

You say not all of them have the flicker...
It is by single color more?
It is by changing the animation speed or by higher speed more flicker?
It is by all different settings for brightness?

On the other way the flicker can be a software PWM issue. Told @Aircoookie about it in the pull request for: Remove analog flicker, but who am i who knows WLED really? ;-) 

Try the tips as first action.

perloo, 2020-02-17:
I had the exactly the same problem with flickering with my WS2801 strips, but I think it's all gone since I added a 220 ohm resistor on the data input.
The problem was very intense when setting low brightness.

karimidze, 2020-03-06:
Thank you so much, guys! The problem was in max. current setting in config - i completely forgot this. 
All the rest setup is ok, except ohm resistor - thanks again for pointing me out.

Concerning Loli1n - i believe that it original, i havent any problem with it yet, markings and pcb are high quality.

Unfortunately, the data exchange speed was not enough for 130 diodes and I had to replace this board onto raspberry pi. I'll try reduce count of diodes and increase data speed.

closed 2020-03-06 by karimidze