wled/WLED #686: PlatformIO no longer compiles

Aircoookie, 2020-02-16:
In a new installation of PlatformIO on Visual Studio (PIO Home 3.1.0, PIO core 4.2.0) compilation of current master fails with this error. No changes have been made which could have caused this. It also affects Travis CI since about 2 days.
```
In file included from .pio\libdeps\nodemcuv2\AsyncTCP_ID1826\src\AsyncTCP.cpp:24:0:
.pio\libdeps\nodemcuv2\AsyncTCP_ID1826\src\AsyncTCP.h:28:33: fatal error: freertos/semphr.h: No such file or directory
     #include "freertos/semphr.h"
                                 ^
compilation terminated.
Archiving .pio\build\nodemcuv2\lib4b6\libIRremoteESP8266.a
*** [.pio\build\nodemcuv2\libc2a\AsyncTCP_ID1826\AsyncTCP.cpp.o] Error 1
```

Removing AsyncTCP library from `platformio.ini` as a test, since it is only used for ESP32 (which earlier versions of PIO seemed to honor automatically) yields a further error, which doesn't make any sense at all:
```
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
xtensa-lx106-elf-g++: error: wled00\wled00.ino.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
*** [.pio\build\nodemcuv2\src\wled00.ino.cpp.o] Error 1
cc1.exe: fatal error: opening output file .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld: No such file or directory
compilation terminated.
*** [.pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld] Error 1
```

On my older, main PIO installation on a different Windows machine (Home 3.1.0 core 4.2.1a3, which sounds like a newer version or alpha, in comparison to the one I installed brand new?) everything works fine with current master.

I would be very grateful for any assistance getting PIO and CI to work again.

fishbone-git, 2020-02-16:
Perhaps linked to platformio 4.2.0.  I'm still on 4.1.0 and do not have build issues.

fishbone-git, 2020-02-16:
error: wled00\wled00.ino.cpp: No such file or directory, is sometimes due to the setting to rebuild autocomplete in the platformio settings.  Disable that, and it should get past that error (if it's the same cause as it was for me)

fishbone-git, 2020-02-16:
vscode updated to pio 4.2.0 this evening, and now it is failing.  This may be something to try:
https://community.platformio.org/t/pio-4-2-0-complile-error-for-asynctcp-id1826/12025

sungla55guy, 2020-02-16:
i also have this issue following

fishbone-git, 2020-02-16:
adding the following works for me for now:
lib_compat_mode = strict
lib_ignore = AsynTCP

Have to add it to any esp8266 env section that you are using.  Probably until Pio 4.2.1 is out.

closed 2020-02-17 by Aircoookie