wled/WLED #687: Static Ip not working

micmax1, 2020-02-16:
I want to setup a static IP but idk how because when I static IP my nodemcu's MAC in my router and fill in this in the WLED wifi config:

Static IP (leave at 0.0.0.0 for DHCP):
192.168.X.X

It doesn't work when saving and rebooting, also when saving&rebooting&re-power my nodemcu, please help (im newbie)

Aircoookie, 2020-02-16:
Hi!
When you set a static IP in WLED, you don't have to set anything in your router config.
If you assign a static IP in the router config, leave the WLED setting at 0.0.0.0, using both is not neccessary and could lead to issues.
Also make sure that if you assign the IP in WLED settings, the gateway IP must also be set to your router IP.

Hope you can get it to work soon!

micmax1, 2020-02-16:
Thnx! It works! I just had to put in my gateway lol

closed 2020-02-16 by micmax1