wled/WLED #689: Every second LED is dark

palengh, 2020-02-17:
My string has every second LED dark when “Disable repeating N LEDs:” is set to “0”.
When set to “1” they’re all dark. Switching to “2” dims out 2 consecutive LEDs

What am I missing out?

version 0.9.0-b1
WS2815b
12V 3.5A supply connected to LEDs

palengh, 2020-02-17:
Rebooting through the app made no progress. 
After cutting the mains, and switching power back on (NodeMCU) has constant power, all LEDs showed up again.

Def3nder, 2020-02-19:
Hi @palengh, could you please look on the segments tab, if there is something entered in the field "spacing" ? "normal" would be "0" for spacing.

palengh, 2020-02-19:
[image]

palengh, 2020-02-19:
As mentioned Above; the “Disable repeating N LEDs:” is set to “0”.

Def3nder, 2020-02-19:
hmmm - I see - the UI for entering the "spacing" value for a segment was just added I think.
So probably there are values in,  but you cannot see the values.

You could either clone the latest master or reset the EEPROM and start over with default values:
```
http://[WLED-IP]/cleareeprom
```

The first would keep the existing values and would suggest to do that 😄

palengh, 2020-02-19:
What do you mean by “ clone the latest master”?

Def3nder, 2020-02-19:
sorry - I thought you would have compiled from source 😢 

Which version did you use (for which ESP device)?
Standard PINs ?

...I will compile a binary for you.

palengh, 2020-02-20:
WLED version 0.9.0-b1
WS2815b
12V 3.5A supply connected to LEDs
Rynning on a NodeMCU.

Def3nder, 2020-02-20:
Hi @palengh,

her we go: NodeMCU version from the actual master:
[WLED_pre-0.9.1_ESP8266.zip](https://github.com/Aircoookie/WLED/files/4229595/WLED_pre-0.9.1_ESP8266.zip)

...after flashing you will see the spacing in the segements tab:
[image]

palengh, 2020-02-20:
Thanks A lot. 
There seems to be an issue with current (not the one you compiled) version. It gets unreachable after 10-50 hours. Power is on and the light switch is turned off. 
Then the NodeMCU is unavailable.

palengh, 2020-02-20:
After flashing:

[image]

Def3nder, 2020-02-20:
hmmm - looks "normal". Is the "every second LED" effect present with the new binary, too ?

palengh, 2020-02-22:
The second led issue seems to be fixed in this latest update. 
Thank you!

closed 2020-02-22 by palengh