wled/WLED #690: E1.31 support ( ws2801, nodemcu )

digitalborderattal, 2020-02-17:
Hello, I run my Ws2801 on a nodemcu with wled. It all works very well. What I haven't been able to do yet is to control wled via E1.31.
Is there a trick? 
Thanks and a big praise. This project is wonderful.

Can I donate something somewhere?

Aircoookie, 2020-02-17:
Hi, I am very happy that you like WLED so much :)
What version of WLED are you using, and what is the E1.31 sending software?
It should work out of the box by configuring the correct IP address of the ESP8266 in unicast mode.

digitalborderattal, 2020-02-17:
I'm using version 0.9.1. I downloaded the master.zip last week, configured it for ws2801 and compiled and flashed it in the Arduino IDE.
I have tried it with jinx, x lights, qlc+ and madmapper with unicast as well  with multicast.

pille, 2020-02-18:
@digitalborderattal: i'm successfully using QLC+4 for controlling my CyLights in multicast mode.
though it took me some time until i got a working setup. make sure you're broadcasting on the correct network interface, in case you have multiple. `tcpdump port 5568` is your friend.

which DMX mode are you using? try `SINGLE_RGB` for the beginning. here's my fixture definition for the DMX modes `SINGLE_RGB`, `SINGLE_DRGB` and `EFFECT`:
[DIY-CyLight90-(WLEDv0.9).qxf.txt](https://github.com/Aircoookie/WLED/files/4221100/DIY-CyLight90-.WLEDv0.9.qxf.txt)
(put under `.qlcplus/fixtures/` and remove the `.txt` extension [github workaround])

OliPassey, 2020-04-15:
@digitalborderattal - when you say you configure for ws2801 - what did you need to do?
Im looking to use this functionality (comfortable making code changes and deploying from arduino ide) but can't find where i need to comment this in. I presume thats how this function is used currently?

OliPassey, 2020-04-15:
Ignore me, found it :)