wled/WLED #691: WLEDAP network disappears when analog polling from A0 on ESP8266.

atuline, 2020-02-17:
On a WeMOS D1 Mini, I added the single line to wled06_usermod.ino, recompiled (0.9.0-b2) and the AP mode WLEDAP network disappeared:

int  micIn = analogRead(A0);

When I remove that line and re-compile, it re-appears.

Even tried disabling things like OTA, ALEXA, BLYNK, CRONIXIE, HUESYNC and INFRARED.

atuline, 2020-02-17:
This can be remediated (at least somewhat), by reducing the polling frequency.

Def3nder, 2020-02-20:
Hi atuline,

"normaly" one loop takes too long if there are WiFi disconnects - you could enable debug-output to Serial Monitor with "#define WLED_DEBUG" (or -D WLED_DEBUG"  in platformio.ini) and check what's displayed as Loop/s.

When WLED is idle this is around 4000 and it should better not drop below three digits.

By the way: I love your FastLED-Demos repo! Some of them have been used in WLED, too 😄

atuline, 2020-02-20:
Thanks, I'll have a look at that. In the meantime, I've now converted my 40 or so displays to WLED and am hosting a lantern making workshop in Vancouver, and each attendee will receive a WLED programmed WeMOS D1 Mini. Am excited to use your platform.

closed 2020-02-20 by atuline

andruby, 2020-10-07:
I've noticed the same problem when I created a usermod that uses a potentiometer as a brightness dimmer. When using analogRead too frequently, it's not possible to connect to the wifi AP.

At 250Hz polling wifi fails. At 10Hz polling wifi succeeds. 

```h
#pragma once
#include "wled.h"

class PontentioBrightness : public Usermod {
  private:
    unsigned long currentTime;
    unsigned long lastTime = 0;
    unsigned long voltage;
    unsigned long lastVoltage = 0;
    const int analogInPin = A0;

  public:
    void setup() {

    }

    void loop() {
      currentTime = millis();
      if (currentTime - lastTime > 100) { // 100ms = 10Hz
        lastTime = currentTime;
        voltage = analogRead(analogInPin);
        if (abs(voltage - lastVoltage) > 4) { // Noise should not trigger a change
          lastVoltage = voltage;
          int newBrightness = int(voltage / (1023.0 / 255.0));
          bri = newBrightness;
          colorUpdated(NOTIFIER_CALL_MODE_DIRECT_CHANGE);
        }
      }
    }
};
```