wled/WLED #692: new effect wont sync with other wled modules

sansillusion, 2020-02-17:
Hi, thanks for the great work ! While trying latest master, I noticed that the new heartbeat effect will not sync with other devices (also updated) and it will not highlight the effect wen clicked in the UI.

sansillusion, 2020-02-17:
After more testing it actually does not work, it activates Ripple Rainbow and does not sync it, if I activate Ripple Rainbow it will sync.

Aircoookie, 2020-02-17:
Oh, sorry! I merged two new effect PRs at the same count and missed to increase the total effect number by one more. Fixed! WLED now officially has over 100 effect modes!

sansillusion, 2020-02-17:
Ok, better but it seems that heartbeat is broken. On my esp8266 with 120 leds it starts somewhere close to the middle (not all leds are used). If I change direction in led preferences it does nothing different (no direction change) and if I change direction in segments it just goes faster. Something seems broken. On my esp32 with 89 leds it goes blank and crashes.

fishbone-git, 2020-02-17:
I get rainbow bands that fade with the brightness of the background colour setting.  intensity slider, which is supposed to control fade seems to make the rainbow wash out to pastel.  It's very odd and doesn't match the code.  Something very odd going on there. I don't know what this effect is supposed to look like as I cannot find video anywhere.

Aircoookie, 2020-02-18:
I can confirm that the heartbeat effect does not seem to work very well. Crashing is a no-go. In my ESP8266 testing only the second half of the segment was illuminated. The `memmove` it uses to manipulate the raw LED buffer is arcane magic (it really isn't, but in WLED we'll need to find a different approach), because this doesn't work for translated segments (it only works if the segment starts at LED 0, is not inverted and has default grouping settings).

@Def3nder for now I replaced Heartbeat with a simpler effect that just modulates the brightness similar to a heartbeat. When I have time, I will take a look at what the original effect from WS2812FX visually looks like and implement it without memmove :)

sansillusion, 2020-02-18:
Thank you that will be awesome. Will compile temporary fix version when I get home tonight.

closed 2020-06-24 by stale[bot]