wled/WLED #695: Tertiary colour fails to sync until other colour is changed

fishbone-git, 2020-02-17:
Seems to have been somewhat recent, although I cannot recall specifically testing (just can't recall ever noticing it).  Tertiary colour won't sync over udp when changed via the ui.  However, once either the primary or secondary is changed, then the tertiary catches up.

Def3nder, 2020-02-20:
Hi @fishbone-git, 

did you refresh the UI after the sync (on the target ESP) ?
...I think the UI will not show other colors without getting a "refresh".

Aircoookie, 2020-02-20:
It actually makes sense that this bug happens. UDP sync is only sent if a value changed.
The tertiary color is the only value that is not checked for changes in the `colorUpdated()` and will thus never trigger a sync packet. I will try and find a way to make the sync work without introducing `colTer` variables just for checking.

pille, 2020-02-20:
@Aircoookie: when you would introduce `colTer`, i could use that in E1.31 `DMX_EFFECT` mode and add a channel for that. i already planned supporting tertiary color in #665, but using `strip.setColor` was slow, so i left it out.

blazoncek, 2022-02-21:
Fixed in recent sync changes (full segment sync).
https://github.com/Aircoookie/WLED/pull/2547 and https://github.com/Aircoookie/WLED/pull/2427
Closing.

closed 2022-02-21 by blazoncek