wled/WLED #696: New version faills to reconnect to wifi if disconnects

davidcreager, 2020-02-19:
Apologies if this is duplicative.   It is mentioned in other issues.
I have three wled drive strips, two on the latest version(0.9.0-b1) and one on the previous version (0.8.6).   When I reboot my router (ie restart wifi) both WLEDs running the latest version do not reconnect until I power cycle them.   The one running the previous version reconnects fine

huggy-d1, 2020-02-19:
If you install the latest build for 0.9.0-b1, does it fix the problem?  The devs found / fixed some wifi related issue(s) fairly recently.

davidcreager, 2020-02-19:
Yikes   it didn't occur to me to try.....I will try now and report back

Def3nder, 2020-02-19:
Hi @davidcreager, there is not new release yet - which device do you use ?
I can create a suiting binary for you from the current master.

davidcreager, 2020-02-19:
> Hi @davidcreager, there is not new release yet - which device do you use ?
> I can create a suiting binary for you from the current master.

Oh that is so kind.   I am using esp8266 devices currently flashed with WLED_0.9.0-b1_ESP8266.bin

snook, 2020-02-20:
There is a large thread regarding wifi connection issues here #644

davidcreager, 2020-02-20:
> There is a large thread regarding wifi connection/reconnection issues here #644

Yes, I read that, however, I thought it worth flagging the different behavior between versions in a separate issue.   Apologies if it is not helpful.

Def3nder, 2020-02-20:
Hi @davidcreager, this is from the the current master branch:
[WLED_pre-0.9.1_ESP8266.zip](https://github.com/Aircoookie/WLED/files/4230368/WLED_pre-0.9.1_ESP8266.zip)

davidcreager, 2020-02-20:
> Hi @davidcreager, this is from the the current master branch:
> WLED_pre-0.9.1_ESP8266.zip

Thank you so much,   I will test this today and provide feedback.

davidcreager, 2020-02-20:
Ok, so I have installed the pre 0.9.1 onto the two strips that were exhibiting the failure to reconnect.   They appear to be working fine.

Before this, I attempted to set up a test to demonstrate the problem and I 'think' I managed to do this in that I recreated the issue, deployed the new firmware which seemed to fix things.   I am hedging my comments a little as it was not as controlled as I would have liked and ideally I would have proved repeatability.

Anyway, I am going to continue with the new firmware and if I have any problems I will post here.

Thanks again for your help.   Very much appreciated

Def3nder, 2020-02-24:
@davidcreager, if resolved could you pls. close the issue ? Thx 😸

davidcreager, 2020-02-24:
One more test and I will -:)   give me a couple of hours!

davidcreager, 2020-02-24:
YIPPEEE!   A good test and it worked perfectly   You guys are the best   Thanks

closed 2020-02-24 by davidcreager