wled/WLED #700: Change gpio0

antonlamers, 2020-02-20:
Hi, I have the quinled digi uno and GPIO0 isn't broken out. How can i change gpio0 (D3)  that I can use a button for the macro’s. D1 and D2 are broken out from the wemos d1 mini to the quinled pcb.
The founder from quinled says compile. But is this possible and how?

Anton

toto79, 2020-02-21:
Yes it is.
Change the line (No. 14) in file NpbWrapper.h and compile your own firmware with platformIO (or Arduino IDE)

Here you'll find a good introduction for platformIO:

https://github.com/toblum/McLighting/wiki/Software-installation

austwhite, 2020-02-23:
I just saw this now just after I made an issue for a feature request for a QuinLED binary.
I am currently doing it by changing the GPIOs in the sketch and manually building.
Make sure you use the GPIO that is pulled high for the button.
On the Dig Uno, two are pulled low (think it is Q1 and Q2) and the other two pulled high.
See https://quinled.info/2019/05/09/quinled-dig-uno-pinoutwiring-guide/
Also, I think Quindor put a request in for the TempSensor to be supported by WLED last year.

closed 2020-06-29 by stale[bot]