wled/WLED #703: No Problem only a question: Control http effects

Will67-1, 2020-02-21:
Control http effectes
Hello,
I am new here and did not quite understand the effects.
How can I switch on the strip with a specific color ,dim it from 0 to 120 and then leave it at these values.

run once effekts ?
a few examples or a link would be helpful

http://192.111.111.111/win&A=60&FX=2&SX=100 ??

Thanks

Olli

Aircoookie, 2020-02-22:
Hi Olli,
welcome! I assume what you are trying to do is to dim it from 0 (off) to 120 brightness slowly.
Try this HTTP command:
`http://192.111.111.111/win&A=120&R=255&G=160&B=0&TT=2000`
Here, A is the brightness you want to arrive at, RGB is the color you want in RGB format and TT is the time you want the dimming/fading transition to last!

I hope I could help you and have fun with WLED :)

Will67-1, 2020-02-24:
Hi,

Yes, thanks.
I still try the different settings and scenes. The project is great. I am happy about the interface settings and many other things that I could never have written in my script (past and copy).
Thank you for your work.

Olli

closed 2020-02-24 by Will67-1