wled/WLED #708: Feature Request: QuinLed Dig Uno GPIO compatible binary

austwhite, 2020-02-23:
Hi,
Firstly thanks for your great work.  I've been using WLED for my holiday lights and been highly recommending it to everyone I know.
I have been using the QuinLED Dig Uno in conjunction with WLED and was wondering if it was possible to have a pre-built binary compatible with this board.  At this stage I am.modifying and building myself, but this can be tedious as my IDE is running on a Raspberry Pi 3.
The QuinLlED Dig Uno uses GPIO 2 / 3 (selectable) for LED data and GPIO 1 for Clock data. (ESP8266)
It also has 4 GPIO's broken out to headers that would be ideal for the IR sensor or button connection, two pulled low, two pulled high.  It also has a Dallas Temp Sensor onboard.
Pinout for esp8266 and ESP32 is listed here : https://quinled.info/2019/05/09/quinled-dig-uno-pinoutwiring-guide/

I know it's a bit of extra work and the TempSensor is not supported yet, but if a version with the GPIOs set to match the breakouts would be great if it was possible.

Aircoookie, 2020-02-23:
Hi, thank you, I am very happy you like it and are even recommending it to others :)

Interesting, from what I heard about the Dig Uno, it doesn't use GPIO 2/3, so I am offering WS2812b binaries for GPIO4 (ESP8266) and GPIO16 (ESP32), but it sounds like you'd like to drive APA102 or similar SPI LEDs. Maybe you mean GPIOs **D**2/3, that would make more sense. Yep, NodeMCU pin numbering is confusing, the D pin number is not the same as the raw ESP8266 GPIO number. Because it doesn't generally make sense to use GPIO1 (it would interfere with serial communication), I would assume you mean pin **D1** (GPIO 5) instead. Here is a binary for APA102 with data pin GPIO4 (D2) and clock pin GPIO5 (D1), hope it is what you need! :)

https://drive.google.com/open?id=15fzgHxXONi_wzrakoqSjXFVq8QBDHnle

austwhite, 2020-02-23:
Thank you for that.
I'll give it a try and see if it is that confusion.
Just to clarify, Quindor does specify TX and RX on his pinout, but it is easy to mix these up I agree.
I'll let you know how it goes.

austwhite, 2020-02-23:
Hi Aircoookie,
The Data line was my mistake.  It is switchable between RX and Wemos pin D4, but the clock is definitely on the TX line on the V2r5 QuinLED Dig Uno board.
D1 and D2 are broken out to pins (pulled low by 10k). D6 and D7 are broken out to pins (pulled high by 10k).
D5 seems to be used by the Dallas DS18B20.

Aircoookie, 2020-02-23:
Hi, no worries! Here is another binary for APA102, using TX for clock and D4 for data: https://drive.google.com/open?id=1zGZf8o5Km7hgced70d56yrKBwtQ4DanW

austwhite, 2020-02-23:
Thanks for your help.
It does make life easier.

closed 2020-02-23 by austwhite