wled/WLED #709: Number of segments

qwert666, 2020-02-23:
Building an infinity mirror cube where each edge would have a separate segment but unfortunately, WLED only supports 10 of these. 

```
#define MAX_NUM_SEGMENTS 10
```

ws2812fx that WLED is based on has it set up for 10 but as this project is targeted for 
ESP8266/ESP32 instead of old Arduino hardware wouldn't it make sense to increase that number a bit - maybe to 30?

Aircoookie, 2020-02-23:
10 is about the maximum that works reliably, but you could certainly try bumping it up a bit (sounds like you'd need 12 for the infinity cube application). The problem here is not the memory used by the Segment objects themselves, but rather the JSON response for the communication between hardware and browser, which can get too long for the sending buffer.

qwert666, 2020-02-23:
Would it make sense to prepare a PR after trying it out? I guess 10 as the limit is just an arbitrary number and with 12 at least it would be applicable for any cube shape objects.

closed 2020-02-23 by qwert666