wled/WLED #711: Using ws2801 leds

CaptainDario, 2020-02-23:
First of all I want to appreciate the awesome software all of you provide for free!
It's has been working great for my ws2812 stripe of leds therefore I wanted to use WLED also for my other ws2801 stripe. But sadly I cannot figure out how to achieve this.
In the releases section I downloaded the "WLED_0.9.1_ESP8266.bin" however I cannot find a way to change the strip type. (Maybe this is not possible from the software?)
Also I do not understand the different releases (is there some page in the wiki which I cannot find?). Do I have to use another version than the "WLED_0.9.1_ESP8266.bin"?
Because I have not seen any other way to use my ws2801 I tried to compile WLED  myself. However I encountered this error:
```
*** [.pio/build/d1_mini/lib1e9] ~/Downloads/WLED/.pio/build/d1_mini/lib1e9: No such file or directory
Assembler messages:
Fatal error: can't create .pio/build/d1_mini/libff1/ESP8266WiFi/WiFiServer.cpp.o: No such file or directory
*** [.pio/build/d1_mini/libff1/ESP8266WiFi/WiFiServer.cpp.o] Error 1
```
and many more errors looking very similar to this one. Is this an error occurring because I use VS Code and not VS?

Thank you very much for your time,
Dario

raoulteeuwen, 2020-02-23:
Hi @CaptainDario . Did you find https://github.com/Aircoookie/WLED/wiki/FAQ and the question on what hw is supported? I see the 2801 is supported, but please check the remark about 2 pins. What controller exactly are you using (Wemos D1 mini)?

CaptainDario, 2020-02-23:
Hello,
thank you for your fast reply!!
Yes, I found the supported section and that my led-chip of choice was supported. But I did not find any advice what the different releases are targeted towards. First I tried the normal 8266 release and afterwards the APA release both without success.
Yes I am using a Wemos D1 mini (or at least some Chinese knockoff from Aliexpress but they always worked great for me). Sorry but unfortunately I do not know which remark you mean. I connected the wires to RX and D4 of the mini.

raoulteeuwen, 2020-02-23:
Hi. I meant the the 2801 is under the section "2 pins (clock CI and data DI) chips:" ... since i do not have one of those strips, i don't exactly know what it means. If you have a Wemos, i think you could try the "WLED_0.9.1_ESP8266.bin" (unless an 2801 needs another and i'm unaware). 

I did find https://github.com/Aircoookie/WLED/issues/529 and https://github.com/Makuna/NeoPixelBus/issues/324 : not sure whether that helps..

'An' issue is how much your controller has. Another thing that influences what bin you need is what pin your config needs. You could check out a https://github.com/Aircoookie/WLED/releases/tag/v0.8.6 version, as afaik, all of those would work with OTA and for devices with only 1MB.

CaptainDario, 2020-02-23:
Yes I looked at it but it only says that my stripe is supported. Maybe my connection is wrong? Is there some document where I can look up which pins need to be connected? I used the two which are mentioned here: https://github.com/Aircoookie/WLED/blob/master/wled00/NpbWrapper.h (line: 7). But maybe I need to use different pins? So where can I find out the correct pins for my stripe?
#529 says "For APA102 2 pin LEDs, the default pins are 0 for clock and 2 for data." is this the right pinout?
Thank you for you effort and time!

raoulteeuwen, 2020-02-23:
Your welcome, but i think i won't be able to help you further. So i hope @Def3nder , @Aircoookie or someone else can

Aircoookie, 2020-02-23:
Thank you @raoulteeuwen ! Yes, indeed, a different binary is needed for WS2801 as the strip type is not selectable in LED settings (yet!).

Here is a binary for WS2801 with clock pin 0 and data pin 2:
https://drive.google.com/open?id=1OWmpU5rsD0bqu7OyyB2Lb876OCcvKlFB

CaptainDario, 2020-02-23:
I will try the given binary in a few minutes.
Just to understand you right, D0 and D2 not gpio0 and gpio2?

Thank you for your help @Aircoookie !!

Aircoookie, 2020-02-24:
It uses gpio0 and 2, so pins D3 and D4. You're welcome :)

CaptainDario, 2020-02-24:
Thank you!
It works perfectly now!
Maybe you could include that a special binary is needed in the supported hardware section. Because I tried a couple of hours to get it working.
Never the less an awesome piece of software you have created!!

Aircoookie, 2020-02-24:
I added the binry to the release!

closed 2020-02-24 by Aircoookie

JSAnyone, 2022-01-11:
Since the special binary isn’t available anymore in the releases, do I have to compile WLED myself or are WS2801 now included in the “main” release?

blazoncek, 2022-01-11:
Almost everything is run-time configurable since April 2020, including WS2801.

Animizio, 2024-02-21:
Sadly the newest version is buggy again.
Its not possible to save the config with gpio0 because "0" is not allowed