wled/WLED #712: Going to Config and Back destroys color wheel

HugoBalder, 2020-02-23:
Hi all,
when i click in the browser on Config and simply go Back, the Colors in the wheel on the main page are gone.
This seems due to the fact, that the Url changed to http://<ip address>/?
When i remove the /? and reload then the Wheel shows all colors again.

Tested on Safari with IOS 13 on a iphone and an iPad Pro.

Not very tragic as i don‘t use the wheel, but just to let you know.

srg74, 2020-02-24:
I experienced same issue but if you using Chrome instead of Safari everything is OK. Maybe its Webkit problem on OS X.

Aircoookie, 2020-02-24:
Hi,
does this happen using the latest version 0.9.1?

HugoBalder, 2020-02-24:
Hi, yes it does. I installed yesterday the latest 0.9.1 release. Today i tried the IOS App and there it also happens whenever you click on the big Back-Button.

srg74, 2020-02-24:
It’s only happening in Safari. My guess it’s Apple WebKit glitch.

closed 2020-07-01 by stale[bot]