wled/WLED #723: ws2812 + Analog White + separate global brightness + MQTT control

marsic-84, 2020-02-25:
I want to use one RGB ws2812 Led Strip and one White Analog Strip, but when i set RGB type in LED setup, the white strip brightness control not available. And when i set RGBW type in LED setup, ws2812 Led Strip becomes uncontrollable and shows strange colors (because it is not sk6812). And i need the ability to control the brightness separately, ideally through MQTT. 
Is it all possible?

huggy-d1, 2020-02-26:
That sounds like two separate WLED µcontrollers, or creating a custom usermod to handle the analog white strip in addition to the RGB strip.

harueg, 2020-02-28:
Have a look at issue #628

marsic-84, 2020-03-03:
> That sounds like two separate WLED µcontrollers, or creating a custom usermod to handle the analog white strip in addition to the RGB strip.

Is there some examples how to use custom usermod??   I need to read mqtt topic with white brightnes data and control one PWM output pin. Can you help me with that?

nelzon101, 2020-06-05:
Hello @marsic-84 

have you find a solution? I would also take one digital and one analog strip on a ESP8266.

closed 2020-10-10 by stale[bot]