wled/WLED #725: My biggest wish

brausepaule, 2020-02-26:
Hello everybody,
is there a way to increase the number of LEDs to 1800?

Everything works fine for me, but I have 300 LEDs more ....

Aircoookie, 2020-02-26:
Hi, yes there is.
See this forum thread: https://wled.discourse.group/t/esp32-flashing/89/5

Note that this is likely to cause issues on ESP8266 due to insufficient memory, but on ESP32 everything will work, just that the pixel refresh rate will be low.

brausepaule, 2020-02-26:
> Hello everybody,
> is there a way to increase the number of LEDs to 1800?
> 
> Everything works fine for me, but I have 300 LEDs more ....

> Hi, yes there is.
> See this forum thread: https://wled.discourse.group/t/esp32-flashing/89/5
> 
> Note that this is likely to cause issues on ESP8266 due to insufficient memory, but on ESP32 everything will work, just that the pixel refresh rate will be low.

Thanks for the quick reply and the link.
I'll give it a try.

thank you very much

brausepaule, 2020-02-26:
> Thanks for the quick reply and the link.
> I'll give it a try.
>
> thank you very much

So, I changed the code accordingly, and everything works very well.

Again thank you very much

closed 2020-02-27 by Aircoookie