wled/WLED #726: Strong Light flash when powerup Ledstrips with power switch

George-Verboven, 2020-02-26:
**Hi,**

I do have a Strong light flash when powerup Ledstrips.
I have this unwanted “effect” with all strips WS2812, WS2815, APA102.

No matter if I use level shifter, or even a capacitor from 1000 uF connected on power and/or 330 Ohm resistor on data line does solve the problem.

After the strong light flash (first couple of LED's) everything works fine.

The unwanted problem doesn’t occur when turning on with the app, but only when turning on the main power. 
(what I prefer because I have the power connected to a Z-wave power switch)

When I use an arduino nano (with another script of course) I don’t have this unwanted “effect” of an Strong light flash when powerup Ledstrips.

So, Isn’t there a way to solve this problem? maybe a delay?
I have no space left for a relay

btw I use a Wemos d1 mini

huggy-d1, 2020-02-27:
On your D1 Mini, which pin are you using for the data pin for your strips?  Does it have a pull up / pull down resistor?  Does it need one?  You can try inserting a resistor between the D1 Mini pin and the data connection, like 470Ω.  Some people have posted this helped them.

George-Verboven, 2020-02-27:
Hi Huggy-d1,
The LED strip is connected to GPIO2 (NodeMCU/Wemos pin D4 ) 
There is no need for pullup/down resistor and I already have inserted a 330Ω resistor between GPIO2 and data from Led strip.

This is how the unwanted effect looks like: [See video](https://youtu.be/tYajrCgg-Z0)

By the way, I also asked the [same question](https://wled.discourse.group/t/strong-light-flash-when-powerup-ledstrips/145) on the Forum from Aircoookie.

mike2nl, 2020-02-29:
@George-Verboven 
the gpio 2 i used while booting.

- D4 GPIO2
- pulled up, 
- HIGH at boot !!!!! about that the light flash
- connected to on-board LED
- boot fails if pulled LOW

Try a different gpio which is not used while booting.
Search for **Pins used during Boot** and **Pins HIGH at Boot** in this document.
[ESP8266 Pinout Reference_ Which GPIO pins should you use_ _ Random Nerd Tutorials.pdf](https://github.com/Aircoookie/WLED/files/4270285/ESP8266.Pinout.Reference_.Which.GPIO.pins.should.you.use_._.Random.Nerd.Tutorials.pdf)

austwhite, 2020-02-29:
I use D1 Mini's and GPIO2 for the data pin, using a QuinLED Dig Uno as the power supply/breakout board and have never experienced the flashing on any strips.
Maybe I am lucky?   
Do you get the flashing on 0.9.1 and 0.8.6?

closed 2020-07-05 by stale[bot]