wled/WLED #727: Wifi Connection

peter67n, 2020-02-26:
Hello, 
I have question, where I have not found an answer yet.

I get the bin file running on an ESP8266 Nodemcu.
On a Wemos D1 ESP8266 I can only see the AP and are able to connect. After setting the network ssid and password, it will not connect. Only AP is available.
On a Wemos D1 ESP32, nothing is working at all. No AP is coming up. 

Would could be the issue, that Wemos D1's are not running?

raoulteeuwen, 2020-02-26:
Hi. Let's exchange a little more info:

- how exactly did you flash? Did you follow the quick start and compile settings at https://github.com/Aircoookie/WLED/wiki ?
- what exact bin are you using? The 0.9.1. from https://github.com/Aircoookie/WLED/releases ? The "WLED_0.9.1_ESP8266.bin" for the D1 8266 and the "WLED_0.9.1_ESP32.bin" for the 32?
- did you try just running the controllers powered via USB without any LEDs attached and see whether you can connect and config them etc?
- how many LEDs are you trying to use?
- what's your power supply?
- can you share your hardware setup including power supply and all wiring?

peter67n, 2020-02-26:
Let me try to provide answers.
Yes I was reading the Wiki and the christmas tree build.
The bin files you are mentioning are exactly the files, I was trying. To flash the ESP's I'm using esptool.py as I do with other software too. 
I have no LED's connected at this time, just the micro to see, if the wifi connection is working and that I have a chance to move further. 
I hope that the software is not really waiting for a connected LED in order to connect to the home network. At least this is my assumption. The nodemcu did everything I have expected at this time. 
My problem is, that I do not understand, why the Wemos D1 parts are not connecting. 
All boards are connected via USB Cable to a MAC for this test. Later I have planned to use a bigger power supply, depending on the amount of LED's. 
Any idea?

raoulteeuwen, 2020-02-27:
I myself am using Wemos D1's next to NodeMCU's, so it is not a generic "Wemos D1's don't work". 

It's interesting that the Wemos 8266 comes up with it's own AP but won't connect to your Wifi. Only things i can think of for that:

- you're not doing anything on the AP/router/network that would prevent a new device to receive an IP (like mac filtering etc)?
- you've tested it while being close to the AP?
- did you try both using DHCP and a fixed IP?
- you're sure ssid and pwd are set correctly?

Not sure whether trying to have it connect to a device in mobile hotspot mode would be a useful test, but maybe try if you can?

It we don't learn anything new from these questions/answers, i hope someone else or one of the devs steps in.

Aircoookie, 2020-02-27:
Hi, sorry you are having issues getting it to work :)

For the Wemos D1, please try @raoulteeuwen 's excellent advice first.

For the ESP32, flashing is a bit trickier. Please try this: https://github.com/Aircoookie/WLED/issues/517#issuecomment-571333712
and let me know if it helps!

peter67n, 2020-02-27:
It's working :-)
Maybe somebody else is having the same problem, so let me note how it was solved.

Wemos D1 ESP8266:
Also the parts were newly bought, I had to erase the flash before the bin file was flashed. Then it worked.
Flash File: WLED_0.9.1_ESP8266.bin

Wemos D1 ESP32:
The issue #517 helped me in solving it. 
Erased the flash once again. Flashed the mentioned boot loader at 0x00000 and after that the bin file at 0x10000. 
Bootloader File: esp32_bootloader.bin
App File: WLED_0.9.1_ESP32.bin

In both cases it connected via dhcp to my router. 
Many thanks for the help

Def3nder, 2020-02-28:
Hi @peter67n,

nice to hear that your WLED is working now - could you please close the issue, if it is resolved ?

closed 2020-02-28 by peter67n