wled/WLED #728: Feature Request: 2D Effects based on segments

stahlfabrik, 2020-02-26:
Hi!

Let me get this out of the way: I love this software!

I currently build a Lamp, which has 8 strips of 12 LEDs each in a circular arrangement. The strips are laid out in a zig zag way.

I use segments now (with the "reverse" option on each second strip) to get each strip to show the same effect. This is very cool. What I miss though is the option to apply effects not only in the one direction of the strip, but also in a second dimension (animating from segment to segment). Thus generating effects like a lighthouse, a classical rotary police signal and so on. Many of the existing effects could be thought in 2D.

I generated a gif to show the
 current 1D behavior, where each segment shows the same. I would love to have 2D effects:-)

Best regards,
Christoph (we could 

[image]

stahlfabrik, 2020-02-26:
Example of the desired effect: https://www.youtube.com/watch?time_continue=7&v=Lymm2JvcB-8&feature=emb_logo

Aircoookie, 2020-02-27:
Hi Christoph,

I'm very glad that you are happy with WLED!

This is a great idea! If I add matrix support (which ist basically just entering width and height and then mapping from an x + y coordinate to the pixel index), this would both open up the possibility of 2D-only effects (2D fireworks, Noise effects, ...) as well as mapping the existing 1D effects to the matrix in unique ways (horizontal or vertical like in the video, but also maybe diagonal and in a diamond shape). In fact, the horizontal mapping is already supported :) If you use the new 0.9.1 version and set the grouping of the segment to the number of LEDs per short vertical strip, it will work just fine :)

German: Ja klar, wir können gerne auch auf Deutsch diskutieren, ich bevorzuge nur meist Englisch weil viel mehr potenzielle Interessenten es verstehen. Deine Wahl :) Viele Grüße aus Hamm!

stahlfabrik, 2020-02-27:
Oh how small is the world:-) my wife is actually from Hamm. She instantly wanted to know what your name is:-) lol

I see absolutely no need to switch to German as you perfectly understood my suggestion for 2D effects. I think clockwise, counterclockwise - horizontally or diagonal, fading from top to bottom. Each variation will have an awesome effect. But I also think that each existing effect could be enhanced by adding a 2D behavior that matches the intend of the effect. 

I understand your suggestion to basically rotate the existing effects 90 degrees by grouping:-) i will try it for the fun:-) but this will not be the same of course as the resolution vertically of the effect gets reduced in my case from 12 to 1:-)

Maybe a matrix mode would indeed be the right mode for this idea. As segments might be used very differently by others. 

Anyway I am happy you like my proposal. If you have anything to share or need help testing or maybe even coding (I cannot promise good help here;-)) let me know!

Best regards from Cologne 

PS some effects like fire flicker are based on their randomness per segment already super nice 2D:-)

stahlfabrik, 2020-02-27:
I tried to change my existing eight segments to be groups of 12 leds to test your proposal. On two of my three tries the wemos rebooted instantly. The third time I had success but then when I changed some effects, like on the third change the wemos rebooted. 

So maybe the grouping code is not so stable yet:-)

Aircoookie, 2020-02-27:
Wow, what a funny coincidence! I'm Christian Schwinne and I live in the Rhynern part of Hamm :)

I will think about the best way to get matrices to work! :)

Sorry you had issues with bootlooping, I think we are somehow running out of memory. I will send you a new binary later tonight, maybe it will be able to fix it!

stahlfabrik, 2020-02-27:
My wife‘s family name is Schwinde. She is from West-Tünnen:-) 

She says she doesn’t know so many people from Rhynern. 

Funny again the similiarity in the name and the closeness of city parts:-)

stahlfabrik, 2020-02-27:
Regarding matrix code: I build a 16x16 matrix I drove by a raspi using python. 

Maybe you could find some interesting ideas for the mapping of x,y coordinates to led indices. Depending on zig zag or not and minding orientation. 

I think most interesting ideas are here: 
https://github.com/stahlfabrik/RibbaPi/blob/master/display/apa102.py

harueg, 2020-02-29:
Hi Everybody
An easy way to achieve this 2D Matrix, would be by implementing the playback from sd-card.
the 2D matrix would be generated in a separat program e.g. Jinx, Glediator or x-lights, either in tpm2 or fseq files.
These programs ( at least Jinx ) leave you any option of wiring the LED strips.

jezmck, 2020-10-18:
Was any headway made on this? It would hugely increase the use cases for WLED.

harueg, 2020-10-18:
Not that I heard of.  Yes it definitely would !

rickerd, 2020-11-30:
I have my matrix of 900 pixels in a horizontal zigzag from top to bottom (50 leds horizontal, 18 strips vertical stacked)
But I want to have vertical animations. The rain now goes from left to right, I want it to be from top to bottom.

Can you help me @Aircoookie @stahlfabrik?

Aircoookie, 2020-12-01:
It is in the pipeline, but not yet started and no timeline yet.
Multistrip support is first, then I might finally get to matrix support :)

rickerd, 2020-12-01:
Don’t rush it. There’s more in life then just code. I’ll wait for the update eventually.
Thanks for the quick reply.

numanair, 2020-12-01:
Would the implementation of this feature be the same as a re-order/rearrange feature? I think I have in mind the same feature request. I didn't do enough thinking before designing my light panel setup and they are slightly out of order. Only noticeable if you look for it though.

Aircoookie, 2020-12-01:
@rickerd thank you!

@numanair kinda, but specifically for matrices (allowing for different layouts, like zigzag, and different starting points), to then be able to run special 2 dimensional effects on them. Custom mapping/re-order is already supported (see line 30 of FX_fcn.cpp), but it is not an adjustable setting, but needs to be set in code and compiled in

numanair, 2020-12-06:
Oh I see that now, thanks. That works exactly as I would hope. Thank you! Looks much better now.

m14socom, 2020-12-12:
This feature would definitely make me build some cool fixtures. I've been thinking about this for along time. There's already a bunch of 2d noise codes written. I was thinking about using one but, that would mean a couple of matrix effects for no more WLED and that wouldn't be worth it.
The ability to add a matrix for 2d noise effects is a absolute must for a future upgrade.  These would make an excellent effect for a wall piece. 
https://github.com/zett42/PlasmaFractal2
Aircookie: You've done fantastic work with this project. It's the best I've seen and is working great for me.

mkarliner, 2020-12-20:
This should be encouragement. https://www.banggood.com/CJMCU-64-Bit-WS2812-5050-RGB-LED-Driver-Development-Board-p-981678.html?akmClientCountry=GB&cur_warehouse=CN
I  just hooked this up with WLED and used a fresnel lens to project it onto the ceiling. 
It's pretty awesome!
Mike

Legsmaniac, 2020-12-20:
> This should be encouragement. https://www.banggood.com/CJMCU-64-Bit-WS2812-5050-RGB-LED-Driver-Development-Board-p-981678.html?akmClientCountry=GB&cur_warehouse=CN
> I just hooked this up with WLED and used a fresnel lens to project it onto the ceiling.

Wow, thanks for that idea.  I just happen to have a few of those 8X8s lying around and I think maybe a lens up the loft somewhere.  And several spare various Arduino boards.  I know what I'm doing tomorrow now.  😂

blazoncek, 2022-07-25:
2D fully implemented and many effects updated in my fork.
Will be available soon in core WLED.

stahlfabrik, 2022-07-25:
Awesome! Looking very much forward to this!

blazoncek, 2022-08-18:
Implemented in #2737

closed 2022-08-18 by blazoncek