wled/WLED #729: Infrared broken in 0.9.1

Vollpanto, 2020-02-26:
I have a 24key remote working perfekt in 0.9.0-b1. After the update to 0.9.1 only brightness works. But when i select a color or an effect on the remote and press then the brightness button the change takes effect.

Aircoookie, 2020-02-27:
Hi! Thank you for the report! It doesn't sound like you have the wrong remote type selected (but you can double check in Sync settings), but rather like the color is not updating correctly...
You can also go to the Segments tab and see if Segment 0 is selected (checked)

@Def3nder do you have any idea what could be causing this? It seems to me like `colorUpdated(NOTIFIER_CALL_MODE_BUTTON)` should be called correctly. Unfortunately I don't have access to an IR setup for testing right now

Def3nder, 2020-02-28:
Hi @Aircoookie, I had a similar issue (before starting to refactor the PIO-ini) and the problem was that WLED_DISABLE_INFRARED was defined.

Aircoookie, 2020-02-28:
Hi @Def3nder ,
you are correct, I wanted to move to having PIO compile releases anyways (since doing it manually in Arduino IDE is too time consuming), and errors like the `WLED_DISABLE_INFRARED` being defined erraniously are less likely to happen. It should be correctly enabled in the release binaries though.

@Vollpanto 's issue seems to be different though in that the remote works, but color changes are not applied until the brightness is changed. Seems like I will have to find my IR sensor to do some testing 😄

Def3nder, 2020-02-28:
Hi @Aircoookie, @Vollpanto, 

I've found the error - the notification was in the wrong scope (one bracket too early).
This way the colorUpdated(NOTIFIER_CALL_MODE_BUTTON) was only called for repeating changes (like brightness) or for the remotes that send code > 0xFF0000.

I've created PR #738 for this.

cnf, 2020-03-07:
Hi, I am on 0.9.1, and my IR doesn't do anything, **unless** when I have preset cycle enabled... Then the remote responds normally. Is this related, or should I create a new ticket?

Aircoookie, 2020-03-08:
@cnf sounds like it is the same issue... try if the brightness buttons work on your remote.

cnf, 2020-03-08:
@Aircoookie I compiled the latest master, and the issue is gone. Very strange it was working with preset cycle enabled :P

closed 2020-05-02 by Aircoookie