wled/WLED #731: Request: ANAVI controller

tpmodding, 2020-02-27:
i am using an anavi controller, and flashed it with WLED, works without issues

it uses a different led pin order

https://github.com/AnaviTechnology/anavi-light-controller
pin order:
https://github.com/AnaviTechnology/anavi-light-controller-sw/blob/91f7bc8f10d9733f647710c24cf57907fd86dcbb/anavi-light-controller-sw/anavi-light-controller-sw.ino#L43
should we add it as device option in nbpwrapper?

Aircoookie, 2020-02-27:
Hi, yes, we could do so :)
Soon I will add a runtime device selector so that it doesn't even need to be changed before compiling!

tpmodding, 2020-02-27:
also ich kann im betrieb zwischen rgb-led oder ws2812b oder apa102 oder xxx wechseln?? rgb leds gibts keine binarys nämlich

Aircoookie, 2020-02-27:
@tpmodding genau, das ist was ich machen moechte. Es gibt langsam so viele Optionen dass man 200 binaries braucht um alle Kombinationen abzudecken :D

closed 2020-07-03 by stale[bot]