wled/WLED #737: more segments

Feger1, 2020-02-28:
Hello!
I have flashed the original ColoLight controller with the WLED Software and it works fine with 10 segments.
For my status display, which I realized with ColoLights, I would need 12 (0-11) segments.

Is there a way to set up more than 10 (0-9) segments?
Can you help me, to find the place in your code, that is responsible for the restriction?

best regards
Feger

Def3nder, 2020-02-28:
Hi @Feger1,

the restriction is due to the heap (memory) of the ESP8266 - you can ry more than 10, but this could lead to reboots.

Change line 50 in `FX.h`:
```
#define MAX_NUM_SEGMENTS 10
```
Be as close as you can to 10.

Feger1, 2020-02-28:
Thank you so much for your answer.
I don`t need any effect, alexa, ...
Please can you tell me how can i save up memory?

huggy-d1, 2020-02-28:
If we were to build our own .bin with the target being a wemos d1 mini with 4MB rather than the targeted 1MB, what changes are required to allow more segments?

Feger1, 2020-02-28:
For explanation:
I send a JSON command from my FHEM to the controller using curl.
The JSON command contains:
- Brightness
- Segment 1-12 with start LED, stop LED, Number of LEDs, segment color

It would not be bad to get the web interface either.
I would not need anything else and could therefore save memory.

Feger1, 2020-02-28:
I have flashed the 0.9.1 with MAX_NUM_SEGMENTS 12
At the moment it works perfectly.
I will check annd test it.

Thank you for your help

Def3nder, 2020-03-01:
@huggy-d1, if I speak from "memory" I mean heap (RAM) not flash - so when using a WeMos D1 mini, you need to use the 4M (flash) version the 1M version will result in the EEPROM being at a wrong position and lossing all you settings.

@Feger1: the memory problem is only when sending JSON answers, so as long as you do not use the JSON API, you will probably not have any issues wioth it.

dtminnova, 2020-03-02:
hi, i have a problem whit segments , on esp12 when i set 3 segments, and try to set effecst, wled delete all the semgents and just led segment 0 ,i have 117 led strip. and ESP8266 NodeMCU CP2102 ESP-12E 
segmends setting was
seg = 0   0to50 leds 
seg= 1    51to100 leds
seg= 2    101to116 leds,
i have updated to 0.9.1, what could be happens, 
 where can i find any user manual for segments on this last firmware?
thanks.

raoulteeuwen, 2020-03-02:
@dtminnova what controller exactly do you have? Since you mention 'esp12', and i found https://www.letscontrolit.com/wiki/index.php/Basics:_ESP8266_Types_and_Boards#Esp-12 , i wonder whether memory could play a role. What WLED bin exactly did you flash? And can you in more detail share the steps you take and at what moment there is a problem? You know only WLED memory location 16 currently saves segments (the rest of the save slots only store 1 segment)

dtminnova, 2020-03-02:
hi thanks for replay, i buy here https://www.amazon.com/-/es/gp/product/B07HF44GBT/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

ESP8266 NodeMCU CP2102 ESP-12E

i upload the WLED_0.9.1_ESP8266.bin
i have 117 led strip , i save on location (favorite 16) , 3 segments, 
seg = 0 , -  0to50 led
seg= 1  -   51to100 led
seg= 2 ---  101to116 leds,
at firts time  i was  having and issue saving the effect on each segments, when i set on the last one the effect then all my segments are deleted, so now i reflash the esp, but now other issue apears, the efects wont save any more ,
what i want  to do is this , save on favorite 1 ( segment 0 -1 effect), save on favorite 2 (segment 1 -second effect, , same on favorite 3 ( segment 2 - other effect,) ,
then on favorite 16,  saves all 3 segments , whit a different effects on each one.
 but nothing happens, if i swich to load every favorite one by one , only the favorite one works , the other are deleted, 
by the way how can i delete or reset every favorite i wont use anymore?
 so i wondering if there is a user manual how to play whit segments step by step,

thanks , mamfred

raoulteeuwen, 2020-03-02:
Hi @dtminnova . Afaik, the only doc about segments can be found at https://github.com/Aircoookie/WLED/wiki/Segments (everyone can improve those pages ;-)).

Not sure if i understand what you're expecting. Am i right in reading your message as: you want to store only a segment of your strip in fav 1, another segment in fav 2, and another in fav 3? Afaik, you can only store one segment, segment 0, in fav 1-15.

If you have problems after flashing (like settings you want to get rid of), one thing to try is first erasing your flash. If you use esptool like recommended, you can do so using "esptool.py erase_flash". Than reflash with the bin you've used.

(getting values to be saved sometimes is a problem, not sure whether the devs have already solved that definitively)

Maindfield, 2020-06-25:
Hi i want it to have 14 segments and i changed the code to get it. But when i save it to slot 16 and reboot the esp only saves 10 segments. Can i make it to save all the 14 segmenta ?

wizardnl, 2020-08-02:
> For explanation:
> I send a JSON command from my FHEM to the controller using curl.
> The JSON command contains:
>
> * Brightness
> * Segment 1-12 with start LED, stop LED, Number of LEDs, segment color
>
> It [...]

could you show me  a json example? i also need to switch on the segments separately

Aircoookie, 2020-12-01:
Max Segment # is now increased to 12 (ESP8266) and 16 (ESP32) in v0.11! You can support even more by increasing the `JSON_BUFFER_SIZE` in const.h. But for ESP8266, it is already close to the stable maximum, so I'd recommend ESP32 if you need an absurd amount of segments :)

closed 2020-12-01 by Aircoookie