wled/WLED #742: E1.31 Issues

jdavis7765, 2020-02-29:
I started having an issue with new FW release. Transmitting E1.31 from LEDFX to my 300 LEDs running on ESP8266 causes issues with what appears to be the last 150 LEDs. It causes them to turn a solid color and only randomly change. I did not have this on my previously compiled version before the PR of the newest version. I did revert back to a compiled bin I had from January 10th with no issues.

jdavis7765, 2020-02-29:
I did find a version I compiled on 2/20 that I confirmed working.

Def3nder, 2020-03-01:
Hi @jdavis7765, so what has been the error ?
Did you find out ?

jdavis7765, 2020-03-01:
> Hi @jdavis7765, so what has been the error ?
> Did you find out ?

I do not receive an error, just the LEDs not working properly. Not sure if it has to do with the new color algorithm with accurate, brightest, etc. settings.

Aircoookie, 2020-03-01:
Thank you for reporting!
This seems to be the same issue as Errurion reported on discord.
I believe it could be either due to the new out-of-sequence packet rejection or the multi-RGB mechanism @pille contributed.

@pille did you try the `DMX_MODE_MULTIPLE_RGB` succeessfully with multiple universes?

I did some testing myself and had no major issues with LEDfx and 300 LEDs. Will do further testing.

pille, 2020-03-01:
yes, currently my test setup is 3 devices in `DMX_MODE_EFFECT`, with one device in `DMX_MODE_MULTIPLE_DRGB` in between (256 pixels span over two universes)

jdavis7765, 2020-03-01:
Let me know if I can do anything to reproduce for you all. Do not have much coding ability at all so can't help fix unfortunately.

pille, 2020-03-01:
@jdavis7765: please describe your setup (devices, universes, addresses, leds)

jdavis7765, 2020-03-01:
Using BTF-LIGHTING SK-6812 (300 LEDs) with ESP-8266. Streaming E1.31 from LEDfx on a hardwired windows 10 computer. Using the default 510 size in LEDfx to transmit to more than 170 LEDs (I think it was 170. Let me know if you need a video or anything. It appears effects work without any issues.

pille, 2020-03-01:
so you're using start `universe=1` and `dmx address=1`?
can you pinpoint the wrong behaving leds? is it at the universe border?
i have no experience with LEDfx.

jdavis7765, 2020-03-01:
Let me flash the new version and see if I can count up to them.

jdavis7765, 2020-03-01:
Looks like it starts roughly at the end of universe 1 (170 LEDs).

jdavis7765, 2020-03-01:
Here is a video of the behavior

https://youtu.be/-xgqLMZvrN8

pille, 2020-03-01:
what's the effect supposed to look like?
it would make sense to put all LEDs in the same color or have a single pixel running light to detect glitches.

jdavis7765, 2020-03-01:
Mirrored on both sides right down the middle. Left is supposed to look like the right. Like I said though, worked fine before the update and if I revert it goes back to normal so that knocks out anything with my network or computer broadcasting the E1.31 data.

pille, 2020-03-02:
i've tested my setup again yesterday evening and everything looked good (v0.9.1).
the animations on my 256px matrix looked ok.
i've also had a look at the config and it's not an easy case, but starts at universe=1 address=23 leaving the last address 512 unused and continues in universe=2.

@jdavis7765: 
- which DMX mode are you using exactly?
- please try 
  - building FW from commit 258def854cb34117914c965eec7c5b52f2b17b59 and test.
  - a simple effect, so one can see the error and its minimal impact. only one channel changing at a time would be ideal (one pixel runner single main color)
  - changing DMX address (increase one by one and test. how does it affect the effect?)

Aircoookie, 2020-03-02:
@pille your results sounds good to me so far. What is your E1.31 sender program by the way? I could imagine that this is an issue isolated to LedFx (it might not handle the sequence byte properly?)

pille, 2020-03-02:
@Aircoookie: i'm using QLC+ 4. you can check for the `seq=` output in `/win`.
with the latest fix (c66af86d88fae5e0b26fd9390ed13f73c53f8769) the effect should rather freeze  than mix up frames (which results mostly in blinking).

jdavis7765, 2020-03-02:
If I have some time tonight I will try to use jinx or xlights to transmit some data. @pille I am not sure of the DMX mode, would that be on the sending program. Not sure I have ever seen this setting.

pille, 2020-03-02:
@jdavis7765: it's a new setting on "sync interfaces" please take the time to check all related settings, too.

jdavis7765, 2020-03-02:
Remoted into home computer, looks like it was set to Multi RGB.

pille, 2020-03-02:
sounds good. that's the backward-compatible default.

jdavis7765, 2020-03-03:
So I did have a chance to mess around for a few mins last night using xlights instead of LEDfx. While I could not get the colors right it seems the effects displayed without the same issue as LEDfx.

Aircoookie, 2020-03-04:
@jdavis7765 were the colors wrong on the entire strip or just beyond the first universe?
Make sure that the DMX address is set to 1 on both and that the universe size is 510.

Seems like the issue is LedFx specific then and not an issue with @pille 's PR in general.
I have yet to look into it, but will do so soon :)

jdavis7765, 2020-03-04:
> @jdavis7765 were the colors wrong on the entire strip or just beyond the first universe?
> Make sure that the DMX address is set to 1 on both and that the universe size is 510.
>
> Seems like the issue [...]

I just think I had something wrong in xlights, not the greatest with that program. The colors were consistently wrong through the strip but no freezing on the 2nd universe of the strip. And it's no problem, I don't use e1.31 often but I just revert to old fw when I do

mjdieter, 2020-03-08:
I'm new here, so please excuse any slip in etiquette. 

I also had a problem when using Wled 0.9.1 version with LedFx. Looking into it I see that the new packet-out-of-sequence check is the culprit. sACN is the library that LedFX uses to send E131 data. It appears to keep a packet sequence number for each universe. If the target Wled controller supports multiple universes (more than 170 pixels), eventually the sequence numbers in each universe diverge and the new sequence check will start ignoring packets since it only keeps one 'lastSequenceNumber', not one for each universe. Packets are ignored, pixels are updated intermittently, dogs and cats start living together, human sacrifice, you know the rest.  

For my part the packet sequence check is not beneficial, I'm just looking for the lights to blink nice, so commenting it out seems to be the quick and dirty fix.

pille, 2020-03-08:
@mjdieter: you're right. i just checked my E1.31 traffic and sequence numbers are separate for each universe, so we should treat them individually. will look into that...

closed 2020-03-10 by Aircoookie

Commander1024, 2020-05-02:
Had the exact same problem, compiled from git - problem resolved. Crazy. :-)

PR #766 (2020-03-10): fixes #742
* track E1.31 seqcuence numbers for each universe individually
* add E1.31 packet out-of-sequence config option (on/off)
  * extends EEPROM format

fixes #742
commit 7b4dc431 (2020-03-10): Merge pull request #766 from pille/E1.31-sequencenumber-per-universe

fixes #742