wled/WLED #749: segments tutorial please

dtminnova, 2020-03-02:
could anyone please explein step by step how to use the segments settings,

 i try to set 3 segments on a 150 led strip, 

save each  segment on a diferent favorite slot whit different solid color each one.

i check  the preset cycle  mark, and use the slot favorite 16 to save all 3segments whit different effect on each one , but when reboot or load each slot then  , only segment 0 works , and the cycle dont work,
 i have check  reboot to presest on config but nothing changes.

this is my strip array
segment 0 set to 0-50 led , saved on slot 1solid color red
segment 1 set to 50-100 led,  saved on slot 2 solid color green
segment 2 set to 100-150 led,  saved on slot 3 solid color blue
segments from 0 to 2 , each one whit a different led secuency saved on slot 16 

thanks in advance ,have a great day
mamfred

Aircoookie, 2020-03-02:
Hi Manfred,
I am sorry, preset slots 1-15 can only save the first segment (Segment 0). This is a technical limitation because I have not yet changed the save structure for presets. Only preset slot 16 can save all segments.
I will do my best to allow for more presets that can store multiple segments soon!

dtminnova, 2020-03-02:
thanks for replay, 
ok now i have made this setup.
segment 0 set to 0-150 led , saved on slot 1solid color red
segment 0 set to 0-150 led, saved on slot 2 solid color green
segment 0 set to 0-150 led, saved on slot 3 solid color blue
segment 0 set to 0-150 led, saved on slot 4 solid color white

then i add 3 more  segments named 1to 3 , 50 leds each one ,whit a different led secuency each segment and  saved on slot 16
then when i check the preset cycle , the 4 slots , 1 to 4 works fine , but secuency stops at the 4 slot 
but the slot 16 dont  work,
 i have set first preset 1 and last preset 16 ,

other question , how can i  delete any favorite slot, in case if i dont want to use that effect  anymore or start from zero, the only way i found is delete the firmware on board.

thanks once again,

Aircoookie, 2020-03-03:
Hi,
first of all, segments are technically not supposed to overlap right now, it could lead to rendering issues. But if it works for your case that's fine :)
When you set the cycle to 1-16, after loading preset 4, it will try to load preset 5, which probably has nothing saved, so it is ignored. Same with all other presets up to 16, so it will appear to "stop" on preset 4 for a long time until finally loading preset 16. I'd recommend you to re-save the effects you have on preset 1-4 to presets 12-15 and set the first preset to 12 and last to 16.

Deleting presets is not possible right now (other than complete reset), but you can always overwrite an existing preset by just saving something else to it. I hope that is not a major issue :)

RAMBORG, 2020-03-07:
A new feature would be great, 
Like preset 1,3,5,10
The other option is the actual mode 1-16

Svrbinek, 2020-03-15:
Hello. I am before deciding to build a stairway lighting at each stage and I would like to use WLED, but I found it has only a limit of 10 segments. But my staircase is 15 degrees. I would like to ask if there will be more segments in the next version?

raoulteeuwen, 2020-03-16:
@Svrbinek please see https://github.com/Aircoookie/WLED/issues/737

ToXIc-Dev, 2020-06-06:
> Hi Manfred,
> I am sorry, preset slots 1-15 can only save the first segment (Segment 0). This is a technical limitation because I have not yet changed the save structure for presets. Only preset slot [...]

Any news on this, would be great to be able to save multiple segments to more than one slot.

closed 2020-10-16 by stale[bot]