wled/WLED #750: Feature request? Is there a way to determine what ESP a WLED is running on?

fribse, 2020-03-02:
Now with the new software the problem of the past is back. I can't remember what ESP I've used in which lights, so how do I found out what it is before I try to update it OTA?
Can it be seen somewhere, or can it be implemented so you choose the right software package???

Aircoookie, 2020-03-02:
Sure, I understand, that might be problematic and I will add this data to a new upcoming info section in the UI :)
You can get some limited info by opening `[WLED ESP IP address]/json/info` in your browser.
Look for `"pin":[2]`, this will tell you the led pin used. Note this only works for WS281x builds, APA102 and analog LEDs will say 2 despite using different pins.
Also, there is an `"arch":"esp8266"` property, it will tell you if the installation is 8266 or 32. Unfortunately it doesn't give you any info to distinguish between 1mb and 4mb flash esp8266 modules.
I hope it will be of a bit of help regardless :)

fribse, 2020-03-03:
Wow, that helps a LOT, thankyou very much. I hope you will include the info in the GUI in future builds, or maybe even have some sort of check that you are using the right firmware (I messed up a esp8266 yesterday by adding a esp32 firmware to it :-o ).

mike2nl, 2020-03-03:
@Aircoookie 
don't forget the wifi strength, haha

closed 2020-07-08 by stale[bot]

Gulpman, 2022-09-25:
Hi, 
I am having the same problem. Although some time has passed since this thread has been closed, my friend google pointed me to it when asking about "wled which esp do i have". 

I'm quite sure I used a version of the ESP-01 but don't know which one.
I am running WLEC 0.13.1 and wanted to upgrade. 

Running /json/info on the IP, I didn't find any pin so that may be a hint. 

From the Info tab in the GUI it says:
```
Build: 2203150
Estimated current: Not calculated
Free heap: 18.6kb
Filesystem: 20/131kB (15%)
Environment: esp8266 2_7_4_7 (1)
```
Is there a way to find out by these informations, which image I should use for upgrade?
If not, is there any other hint, how I can find out what hardware I used without opening the box?

If the answer to both of the above is no: It would be great if you could add some info to the Info tab which helps a "one year later user" ;) to determine which version needs to be used for upgrade.

It may be possible that OTA is not possible as stated here: https://kno.wled.ge/basics/install-binary/#flashing-method-4-ota-update but the web interface under Config-> Security & Updates shows me all the information to do a Manual OTA Update.

Thank you very much in advance four your help.

softhack007, 2022-09-25:
Actually this information is available in the 'EspClass ESP' object of arduino-esp32, I think a similar class also exists for ESP8266. For 8266, it might be more important to know the flash size. For ESP32 I've created a small arduino sketch that shows how to basicially get the information, see second half in https://github.com/atuline/WLED/issues/231#issuecomment-1251292280

Aircoookie, 2022-09-26:
In your case, filesystem size is 131kB, so you have originally used a 1MB flash binary (the 4MB binary has a filesystem size of 983kB)