wled/WLED #752: Touch buttons integration on ESP32

Uriziel01, 2020-03-03:
Sorry for the naive approach but I do not have much time to dig into this for now :)

Can someone please point me to the direction where I could integrate functionality (on ESP32) of a touchRead() on three buttons where first toggles on/off segment 0, second toggles segment 1 and last toggles both of them. Is there better place a than `wled09_button.ino`?

Aircoookie, 2020-03-04:
Hi, no worries :)
The best place for custom code is the `wled06_usermod.ino`. It allows you to more easily integrate your code and even share it with others if you like.

Since you want to toggle segments, keep in mind there is no official mechanism for that in place yet.
This should work though:

```cpp
uint16_t lastStop[2];

void toggleSegment(byte segm, bool on)
{
  WS2812FX::Segment& seg = strip.getSegment(segm);
  if (on) {
    strip.setSegment(segm, seg.start, lastStop[segm], seg.grouping, seg.spacing);
  } else {
    lastStop[segm] = seg.stop;
    strip.setSegment(segm, seg.start, 0, seg.grouping, seg.spacing);
  }
}
```

Uriziel01, 2020-03-06:
Nice, thanks for such a detailed explanation!
Will try it as soon as I get all the ordered parts :)

ToXIc-Dev, 2020-07-10:
Could this be added as a feature of WLED @Aircoookie ?

Aircoookie, 2020-07-10:
Sure, especially given that the ESP32 supports touch without additional hardware!

RAMBORG, 2020-09-09:
Hope you can integrate the touch function (esp32).

For first integration touch on or off.

Next maybe with long hold dimm function.

donabi, 2020-09-14:
That would be realy nice. I've build a "cloud-looking-hexagon"-thing with lots of screws.
And i realy would like to use the screws as touch-buttons:
[image]

On/Off, Next Favourite, Dimm, Speed, Start Night-Timer ... and so on. There are 10 buttons to use :)

My Project (german):
https://www.maker-forum.net/forum/index.php?thread/161-mein-erstes-wled-projekt/&postID=704#post704

Aircoookie, 2020-09-20:
Thanks to @Jakeler 's PR #1190 this is now included! (you need to compile with the `TOUCHPIN T0` flag)

@donabi that looks amazing!

Monacoslo, 2020-12-05:
> Thanks to @Jakeler 's PR #1190 this is now included! (you need to compile with the `TOUCHPIN T0` flag)
> 
> @donabi that looks amazing!

Can you please help explaining how can I now use touch sensors as button alternative on ESP32? What exactly needs to be done? 

I use WLED_0.11.0_ESP32.bin with esp32-wroom module

Jakeler, 2020-12-06:
@Monacoslo
You need to compile it from source, follow these instructions: https://github.com/Aircoookie/WLED/wiki/Compiling-WLED
Set `default_envs = custom32_TOUCHPIN_T0`.

You can also scroll down and edit the `[env:custom32_TOUCHPIN_T0]` or any other definition, notice that build_flags includes the `TOUCHPIN` flag. Default is T0 (= GPIO4), but T1, T2... up to T9 is possible.

Monacoslo, 2020-12-06:
Thanks for answer! But unfortunally I can not compile in VSC (latest).
After hitting checkmark, I get
`> Executing task: k:\Arduino\WLED\WLED-0.11.0\pio.exe run <

The terminal process failed to launch: Path to shell executable "k:\Arduino\WLED\WLED-0.11.0\pio.exe" does not exist.

Terminal will be reused by tasks, press any key to close it.`

Monacoslo, 2020-12-06:
Since I have also seen already a few questions on facebook regarding this option, maybe it would be great if you also add the compiled option to downloads list :)

blazoncek, 2022-07-25:
Touch inputs on ESP32 are fully runtime implemented since 0.13 (provided you select correct, touch enabled pin). Closing.

closed 2022-07-25 by blazoncek