wled/WLED #753: Changing Led count doesn't work above 10

ozzi91, 2020-03-03:
Hello!
 I have strange problem. Every setting like turning on, setting colors works only for first 10 Leds. After changing number of Leds above 10 it has no effect. Number is changing but turning on or off, setting effects colors etc works only for first 10 Leds. I know the rest of Leds work fine because sometimes after nodemcu restart they are turning on. Sometimes I can change colors for every led but it works only fo mby 10 sec max. After that sometimes I lost connection to device it looks like restart.

My setup is nodemcu, 300 Leds with external 5v 10A power supply. Connected like on drzzs video. Do u know what can be wrong? What can I check?

huggy-d1, 2020-03-04:
Do you have multiple power injection points?

raoulteeuwen, 2020-03-04:
What kind of LED-strip is it? The strips with '3 colors in 1 pixel' take up till .06 A/pixel, so 300 would need 18 A power afaik. It could also be a data line damage between your 10th and 11th LED?

Aircoookie, 2020-03-04:
I would also tend to say that this might be a hardware issue with your 10th or 11th LED. If possible, try a different strip to see if it works beyond 10 LEDs. Or try installing a simple test sketch for LEDs to make sure your hardware is good and it is actually a WLED problem :)

ozzi91, 2020-03-04:
Hmm. It seems like it is hardware issue with led. strange thing. They didnt work even one season and without touching they went dead :/ It seems i need to cut one long strip and check everyone led :/

huggy-d1, 2020-03-05:
Have you confirmed the problem is hardware?

ozzi91, 2020-03-08:
Yep, i tried a different nodemcus, including wemos d1 mini aswell, even another software and it is not working :(

raoulteeuwen, 2020-03-08:
But did you also try another LED strip? I understood we all assumed the most likely cause was a problem between LED 10 and the rest

ozzi91, 2020-03-22:
Ye i tried another strip and it works :/  what could damage strip? Any ideas? How can i use next leds? I just dont want to loose all leds now :/

raoulteeuwen, 2020-03-22:
You could either try to repeair the traces between the last working and the first non-working led by soldering a wire for every trace, or cut of the 1st 10 leds and start from the 11th?

ozzi91, 2020-03-22:
Could you tell me if i am wrong or not. I disconnected cables from strip start point. After that I connected power, gnd and cable signal to 12 led without cutting strip. Should they work like this? Or i need to cut first anyway? I did without soldering by pushing jumper cable and leds didnt work, they go on for like 0,1 sec and turned off.

raoulteeuwen, 2020-03-22:
It should work if the connections are o.k. Afaik, the power is still distributed to the 1st 10 LEDs also, if the power traces are in tact, but the data is not flowing 'backwards', so if all other things would be fine, you would see LED 12 and later turn on.

I read back this thread. In the beginning you state "I know the rest of Leds work fine because sometimes after nodemcu restart they are turning on. Sometimes I can change colors for every led but it works only fo mby 10 sec max. After that sometimes I lost connection to device it looks like restart". So even the LEDs after the 1st 10 sometimes light up? Any possibility it is the power supply? Do you have either another (and more powerfull) power supply or a LED-strip with less LEDs?

ozzi91, 2020-03-22:
Hmm so it is not working properly at all. Mby my connection are not ok, coz i didnt solder it, just want to check it out frist before i cut strips down.

Now im using fastled library from arduino to exclude software problem. And like on wled only first 10 leds are on. I tried 2 different power supply and no difference. Now i even try to launch it by powering from nodemcu pin and the same result. Now my goal is to retreive as much leds from that strip as i can :)

ozzi91, 2020-03-25:
New news!

I cutted down 10th led, After that i loaded a simpel sketch from arduino, everyone led was working again. After that i flashed wled on NEW nodemcu. Everything was working pefect. I started to have fun with it, and changed effects very often. After a few effects leds restarted.  After that frist 12 leds are changing due to changes on app. It is like second time it happeneds to me... do u know what is wrong? when i cut power off and turn it on, every led is going on on default colour, but modes are changing only for first 12 leds.

raoulteeuwen, 2020-03-26:
Well, based on the info so far:

- you said it did work (even after your last change with cutting off the 1st 10 LEDs or so) for while
- you said another LED strip did work fine (how many LEDs were on that strip?)
- you never reacted to my point your power supply seemed underpowered to the nr of LEDs

i would either think it is a bad quality LED strip or i would try a higher spec'd power supply (or strip with less LEDs).

ozzi91, 2020-04-14:
Hey guys! New info! I added a powersupply on every 1m of led strip. Now after a disconnecting plug and connecting it again wled is working very nice, but after a 10min or so (random time) it starts working on first 10-15 leds. But it seems power is still on on another leds coz they are standing on last rremebered state. Im very very disapointed, that ledstrip makes me crazy. I am thinking of throwing it away:D I changed power supplies and 0 differences

raoulteeuwen, 2020-04-15:
Thanks for reporting back. Really difficult to understand what is the problem. Sounds like it would be great if you can try with another LED strip, and maybe consider buying another brand this time?