wled/WLED #757: Alexa capabilities => can we control light effect too?

sommo, 2020-03-05:
Hi!
amazing project, i'm really enjoy it!
i tried the alexa voice pairing and it works ok, alexa discover it and i can turn in on/off, change color and brightness via voice.
Is there a way to control effects too?
Thank you!

huggy-d1, 2020-03-05:
The github source referenced for the Phillips Hue emulation provides a "color loop" effect, or no effect option.  That effect is not supported by the Alexa / Phillips Hue emulation in WLED.

I agree it would be fantastic if I could say "Alexa, enable effect on Party Strip" and it execute preset 0 (or an Alexa specific preset) and it figures out what to do.

Of course, it might already be in there somewhere else and I missed it in the code.

Aircoookie, 2020-03-07:
Yes, unfortunately it is not directly possible to set an effect via Alexa.

It is possible to set WLED macros that get executed on on/off command. This is not so useful though, because you lose on/off control, but it works nicely if you want the light to apply a certain preset every time you turn it on via Alexa, for example.

ricardo-brenner, 2020-03-12:
You can use some of the colors to control the effects, for example, when you ask alexa to change color to blue, it set the lightning effect

closed 2020-07-18 by stale[bot]

tablatronix, 2020-12-06:
@ricardo-brenner I was thinking the same, alexa can send colors but how would you map them?

jegaha, 2021-04-03:
Hey guys,
I was thinking about a prefix selection solution too and came across the misusage of the percentage for the brightness invocation from Alexa.
‚Alexa set brightnes to 1 percent‘ selects preset 1
So eg. a brightness from 1 to 10 selects a prefix and the values above adjusts the real brightness.
Yes, as a tradeoff a little bit of brightness solution is lost but you get an easy to remember preset mapping.

siggel, 2022-02-02:
Just to inform the participants of this issue, that I made a proposal for a solution here:
https://wled.discourse.group/t/call-different-presets-from-alexa/306/5