wled/WLED #758: WLED_0.9.1_ESP8266_1M_full.bin reboots when attempting to connect to wifi router

sfranzyshen, 2020-03-05:
I downloaded th WLED_0.9.1_ESP8266_1M_full.bin firmware and installed it on a 1MB esp8266 ... it functions fine until I try and connect to my wifi router ... then it repeatedly reboots with the following exception ...
```
Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
>>>stack>>>
ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 01a0
3ffffed0:  00000000 00000001 3fff1f34 3fff2398  
3ffffee0:  3fffff40 3fffff4c 00000000 402228e3  
3ffffef0:  00000050 00000005 3fffff20 3fff2170  
3fffff00:  00000003 3fff2378 3fff184c 3fffff4c  
3fffff10:  3fffff40 00000050 3fff2398 4022294e  
3fffff20:  3ffeabd3 3fffff40 3ffeabcf 3fff2170  
3fffff30:  3ffeabd3 3ffeabcf 3fff2398 4021c07e  
3fffff40:  00706374 3e01a8c0 83c6a7f0 64656c77  
3fffff50:  3fff5500 842210bc 40100b30 3fff2170  
3fffff60:  3fff2175 3fff216f 00000000 4021c2de  
3fffff70:  00000000 402210bc 3fff2598 3fff25d8  
3fffff80:  00000000 00000000 00000001 3fff25d8  
3fffff90:  3fffdad0 00000000 3fff2598 4021f3e7  
3fffffa0:  feefeffe feefeffe feefeffe 402304c4  
3fffffb0:  feefeffe feefeffe 3ffe8800 4010089d  
<<<stack<<<
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4000bdc8
EXCVADDR: 0x00000000
Decoding stack results
0x402228e3: umm_info(void*, int) at /home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 1062
0x4022294e: umm_info(void*, int) at /home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 1080

```

sfranzyshen, 2020-03-05:
same results with WLED_0.9.0-b1_ESP8266_1M_full.bin ...

huggy-d1, 2020-03-06:
I have no idea if there even exists such a thing as a 512k esp8266, but if it does, maybe you have one??

Have you tried erasing your 8266?

What memory size does your esp8266 flashing tool say it has?

austwhite, 2020-03-06:
There are two versions of ESP-01 boards.  The "black" one is 1Mbyte, the other is 512K.
I would not recommend WLED on the 512k as you will lose so many features.
There use to be a very stripped down 512k binary though, but you cannot use OTA on the 512 at all.

sfranzyshen, 2020-03-06:
I have the black 1m and can verify the flash chip is 1m not the 512 blue board ... also the 1m-ota works just fine ...

huggy-d1, 2020-03-06:
When you connect to your 8266 with your favorite flashing tool, what details does it share when it first connects to the device?

What does it do when you perform a full erase before flashing the 1m full bin?  Does it say it succeeds and if so, does it feel normal while erasing?

sfranzyshen, 2020-03-07:
```
python3 .\esptool.py flash_id
esptool.py v2.6
Found 3 serial ports
Serial port COM6
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:8e:83:16
Uploading stub...
Running stub...
Stub running...
Manufacturer: e0
Device: 4014
Detected flash size: 1MB
Hard resetting via RTS pin...
```
**WLED_0.9.1_ESP8266_1M_full.bin** flashes (with full erase in DIO mode), boots, and runs fine ... until I try and connect to my wifi router then it continually reboots (with the exception above) ... I have to re-flash it to fix it. Please don't overlook that it flashes and works ... and keep being fixed on it being a hardware issue (**it's not a 512K board**)

huggy-d1, 2020-03-07:
Actually, I believed it from the beginning. I was just checking to make sure your flashing tool was properly identifying and erasing the permanent settings stored in flash. After all, when you save your wifi connection details, wled saves them to flash memory and then restarts and then it reads from the flash memory locations it thinks it wrote to. If the flashing tool setup the firmware incorrectly during install, I don’t know what it would do, but it probably isn’t what you would want. 

Hope that helps you understand where my thinking was. There’s the reality you have, there’s the reality the flashing tool has, and the reality WLED has. When all match, happiness happens.

Please try a different flashing tool. I know it sounds ridiculous, but unless your wifi has unicode characters or other characters that could confuse WLED, it might just be the flashing tool not working perfectly. 

I’ve heard people have problems with esp8266’s and fixed them by using a different flashing tool.

raoulteeuwen, 2020-03-07:
@sfranzyshen could you please try to toggle some settings and see whether it makes a difference:

- switch off/on mDNS
- switch off/on NTP
- switch off/on MQTT

(and if those don't make any difference, switch off/on other settings: UDP, E1.31, Hue, Alexa...)

Def3nder, 2020-03-16:
Hi @sfranzyshen , sometimes it helps when you use Arduino IDE and compile a simple sketch that uses WiFi and stores WiFi credentials (or just flash Tasmota or ESPurna to it).

Then do a complete erase either with ESPtool or with the ESP8266Flasher (or with PlatformIO) and re-flash WLED.

toto79, 2020-03-21:
I could solve a same issue (reboot after wifi connect) by the following changes in platformio.ini 
  
  #-D _IR_ENABLE_DEFAULT_=false 
  #-D DECODE_HASH=true 
  #-D DECODE_NEC=true
  #-D DECODE_SONY=true 
  #-D DECODE_SAMSUNG=true
  #-D DECODE_LG=true

but i don't know if it's the root cause, because i need & made more changes in npbWrapper.h, ( #define IR_PIN -1, use ws2811, ...  nevertheless my issue was solved.

SabreFrenzy, 2020-07-15:
I too am having an issue with the unit rebooting once i enter my WiFi details.  I am using a wemos Miniesp-8266:

Flashed with multiple tools:

Serial port COM8
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 2c:3a:e8:4c:1e:84
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.8s
Flash params set to 0x0040
Compressed 544816 bytes to 375621...
Wrote 544816 bytes (375621 compressed) at 0x00000000 in 33.2 seconds (effective 131.4 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.

Works fine in AP mode.  Once WiFi Details entered i receive the following:

19:11:53.733 -> 
19:11:53.733 -> Exception (28):
19:11:53.733 -> epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
19:11:53.767 -> 
19:11:53.767 -> >>>stack>>>
19:11:53.767 -> 
19:11:53.767 -> ctx: cont
19:11:53.767 -> sp: 3ffffd20 end: 3fffffc0 offset: 01a0
19:11:53.767 -> 3ffffec0:  00000000 3fffc6fc 00000050 3fff1c34  
19:11:53.767 -> 3ffffed0:  3fffff30 3fffff3c 00000000 4022b297  
19:11:53.767 -> 3ffffee0:  00000050 00000000 00000000 3fff19ff  
19:11:53.767 -> 3ffffef0:  00000003 3fff1be0 3fff10d0 3fffff3c  
19:11:53.767 -> 3fffff00:  3fffff30 00000050 3fff1c34 4022b302  
19:11:53.767 -> 3fffff10:  3ffea507 3fffff30 3ffea503 3fff19ff  
19:11:53.801 -> 3fffff20:  3ffea507 3ffea503 3fff1c34 4021cf34  
19:11:53.801 -> 3fffff30:  00706374 146ba8c0 83c6a7f0 64656c77  
19:11:53.801 -> 3fffff40:  3fff4e00 84223838 40100ae4 3fff1e2c  
19:11:53.801 -> 3fffff50:  3fff1a05 00000000 3fff0ce0 4021d1b2  
19:11:53.801 -> 3fffff60:  00000000 d9a3a378 3fff0c00 3fff1e2c  
19:11:53.801 -> 3fffff70:  00000000 00000000 3fff0ce0 3fff1e2c  
19:11:53.801 -> 3fffff80:  3fffdad0 00000000 3fff0ce0 4021d221  
19:11:53.836 -> 3fffff90:  3fffdad0 00000000 3fff1dec 4021db4c  
19:11:53.836 -> 3fffffa0:  feefeffe feefeffe feefeffe 4023209c  
19:11:53.836 -> 3fffffb0:  feefeffe feefeffe 3ffe8800 40100851  
19:11:53.836 -> <<<stack<<<
19:11:53.836 -> 
19:11:53.836 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,0)
19:11:53.836 -> 
19:11:53.836 -> load 0x4010f000, len 1392, room 16 
19:11:53.836 -> tail 0
19:11:53.871 -> chksum 0xd0
19:11:53.871 -> csum 0xd0
19:11:53.871 -> v3d128e5c
19:11:53.871 -> ~ld
19:11:53.941 -> Ada
19:11:59.985 -> 
19:11:59.985 -> Exception (28):
19:11:59.985 -> epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
19:11:59.985 -> 
19:11:59.985 -> >>>stack>>>
19:11:59.985 -> 
19:11:59.985 -> ctx: cont
19:11:59.985 -> sp: 3ffffd20 end: 3fffffc0 offset: 01a0
19:11:59.985 -> 3ffffec0:  00000000 3fffc6fc 00000000 3fff1c34  
19:11:59.985 -> 3ffffed0:  3fffff30 3fffff3c 00000000 4022b297  
19:11:59.985 -> 3ffffee0:  00000050 000045ff 3fff1e2c 3fff19ff  
19:11:59.985 -> 3ffffef0:  00000003 3fff1be0 3fff10d0 3fffff3c  
19:12:00.020 -> 3fffff00:  3fffff30 00000050 3fff1c34 4022b302  
19:12:00.020 -> 3fffff10:  3ffea507 3fffff30 3ffea503 3fff19ff  
19:12:00.020 -> 3fffff20:  3ffea507 3ffea503 3fff1c34 4021cf34  
19:12:00.020 -> 3fffff30:  00706374 146ba8c0 83c6a7f0 64656c77  
19:12:00.020 -> 3fffff40:  3fff4e00 84223838 40100ae4 3fff1e2c  
19:12:00.020 -> 3fffff50:  3fff1a05 00000000 3fff0ce0 4021d1b2  
19:12:00.020 -> 3fffff60:  00000000 fb7f259d 3fff0c00 3fff1e2c  
19:12:00.055 -> 3fffff70:  00000000 00000000 3fff0ce0 3fff1e2c  
19:12:00.055 -> 3fffff80:  3fffdad0 00000000 3fff0ce0 4021d221  
19:12:00.055 -> 3fffff90:  3fffdad0 00000000 3fff1dec 4021db4c  
19:12:00.055 -> 3fffffa0:  feefeffe feefeffe feefeffe 4023209c  
19:12:00.055 -> 3fffffb0:  feefeffe feefeffe 3ffe8800 40100851  
19:12:00.055 -> <<<stack<<<
19:12:00.055 -> 
19:12:00.055 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,0)
19:12:00.089 -> 
19:12:00.089 -> load 0x4010f000, len 1392, room 16 
19:12:00.089 -> tail 0
19:12:00.089 -> chksum 0xd0
19:12:00.089 -> csum 0xd0
19:12:00.089 -> v3d128e5c
19:12:00.089 -> ~ld
19:12:00.155 -> Ada

This is when i flash WLED_0.10.0_ESP8266_1M_full (1).bin

If i flash WLED_0.9.0-b1_ESP8266.bin it works fine but i want segment control in HA.

SabreFrenzy, 2020-07-15:
using WLED_0.10.0_ESP8266.bin works... 

Any reason why WLED_0.10.0_ESP8266_1M_full (1).bin would not load correctly?

toto79, 2020-07-16:
see Issue #1030 and #1025

closed 2020-11-21 by stale[bot]