wled/WLED #760: More button action idea

sommo, 2020-03-07:
Hi! Awesome project! (I could not start a new topic without saying it).
It could be possible to add more action to the button input?
For now you can press it, press it double and long press.
It could be usefull to have also 3-click, 4-click and maybe other combination, like click-long click. So for example you can use this setup:
1click=change effect
2click=change color
3click=Toggle on/off
Long press=Brightness+ (Until release)
Click+long press=Brightness-

Another question:
Is there a macro that can do this:
Long press=brightness+ to max and if you keep pressed once on max birghtness start decreasing brightness and so on? Like a loop from min bri to max and vice versa until release the button

Thank you

Aircoookie, 2020-03-07:
Hi, thank you!
I like the idea of adding more functions to the button with more than double presses.
Especially brightness control, as that isn't easily possible. I will implement such a feature soon. I think it would make sense to change the default long press action when no macro is set to fade up and down the brightness until the button is released :)

damveld, 2020-03-07:
Love your work!! really awesome
I came here to ask exactly the same.
- could you expand the double press to like six times press..
- enable cycling through brightness with long press (like other single button light controls) 
(and it will start in the opposite direction compared to the last press). I could not find any macro for that.

Thank you for all the great work

sommo, 2020-03-08:
Could be also useful add more input button, if you could add other 2 input, we could add a bri+ and bri- button. Or do other other stuff.
Mapping input button (like a variable) to the gpio pin, via web it will also be very useful and smart, so I can choice button and pin.. for example.. button1 => D3, button2 => D4.. and so one.. and select what action/macro will be execute on click, long press, double click, click+long press.... of every single button..
But this could be a long work for coding..

Inviato da iPhone

Aircoookie, 2020-03-08:
Yep, multi button support is also cool and something planned! Keep in mind you'll have to add a 10k pullup resistor to the all button pins but gpio 0 :)

mascolo74, 2020-04-08:
sorry my english,
I use wled to illuminate the outdoor patio, white light, but then at Christmas, I convert the lights to make effects.
it would be nice to be able to insert a button for ignition and start-up, as is done with a home light bulb, I tried but I can't, does anyone have any suggestions?
congratulations wled for me the best...

raoulteeuwen, 2020-04-08:
Hi @mascolo74 you're replying to a closed ticket; i personally don't mind, but not sure what the etiquette is. Anyway

- in the quick start at https://github.com/Aircoookie/WLED/wiki it says "Optionally connect a normally open pushbutton to GPIO0 (NodeMCU/Wemos pin D3) and ground."
- if that doesn't work, maybe https://github.com/Aircoookie/WLED/issues/199#issuecomment-528067206 does (also see the other info in the thread)

RAMBORG, 2020-09-06:
really nice :-) 
finde super das du immer weiter machst!!!!!

strong powerful Aircoookie :-)

mwdsonny, 2021-03-23:
My brother in law is making a display case, and the wifi functionality works great for his use, I thing he would like it to do is be able to have a normally open (or closed) switch that the case lid keeps pressed, that upon opening would run a light show, and then fade over 30 seconds, or a preset time. Is any of the fuctionality to run while switch is on built in?

blazoncek, 2021-10-03:
Most of the features supported with multibutton support added in 0.13. Closing.

closed 2021-10-03 by blazoncek

mistic100, 2021-12-26:
Was the dimming feature implemented ? I am looking to use a single button like OP suggested : "Long press=brightness+ to max and if you keep pressed once on max birghtness start decreasing brightness and so on"

Thanks

blazoncek, 2021-12-26:
Use button > 0 for that.

mistic100, 2021-12-26:
> Use button > 0 for that.

Can you be more specific ? Do you refer the number of the button (0 or 1 on the ESP8266) ? or the GPIO pin ? or the macro command ?

blazoncek, 2021-12-26:
ID

Mopfus, 2022-03-26:
Was not sure where to post my question. So just for the interconnection, you can look here at issue #199

Mopfus, 2022-03-26:
> Hi, thank you! I like the idea of adding more functions to the button with more than double presses. Especially brightness control, as that isn't easily possible. I will implement such a feature [...]

Was not sure where to post my question. So just for the interconnection, you can look here at issue #199

blazoncek, 2022-03-26:
You can't unless you rewrite button logic in source.

evazzoler, 2022-12-02:
The function in the video shown by @Mopfus is basically the default working mode for every dimmer in the world (KNX, analogic, etc.). In other words every dimmer can be used with a single button: short press toggle on/off, long press dimmers. Developing a different working way is a mistake both for an hardware producer and for a wled firmware developer. Please @Aircoookie, standardize wled dimming function!

Mopfus, 2022-12-02:
> The function in the video shown by @Mopfus is basically the default working mode for every dimmer in the world (KNX, analogic, etc.). In other words every dimmer can be used with a single button: [...]

Thats true @evazzoler . I am still waiting for this possibility. Until then I have to use an ESP32 with ESPhome on HomeAssistant and have to program everything by my own. It works. But the other really awesome features of WLED are missing.

blazoncek, 2022-12-02:
If you use button (with ID) 1 you get a ramping brightness using long press.
@evazzoler please provide a PR with a better (standardised) solution without hampering other functions or bloating the code.

evazzoler, 2022-12-02:
> Thats true @evazzoler . I am still waiting for this possibility. Until then I have to use an ESP32 with ESPhome on HomeAssistant and have to program everything by my own. It works. But the other [...]

I still don't have HA (I will) but want to leave wled wireless-independent with its own buttons that always works. It is at my kitchen working area and my wife and childrens have to use it simply and in case the AP connection falls down... :-)

evazzoler, 2022-12-02:
> If you use button (with ID) 1 you get a ramping brightness using long press. @evazzoler please provide a PR with a better (standardised) solution without hampering other functions or bloating the code.

I tried both B0 and B1 with values leave to zero. Upon in time (last summer?) this function was working on B0 as expected. Now I'm going to production with latest release and B0 long pressend change to random color, plus puts wled in AP mode after 5 seconds (too few!). B1 with values leave to zero does nothing.

Ok for using B1 for definitely avoiding wled to go accidentally in AP mode (mind at a wife/children-sensible device). The standard function should be:

- Short press: toggle on/off

- Long press: while pressing it changes brightness in a loop 0->255->0->255->0->255->... so when it reaches zero start to increasing, when it reaches 255 start decreasing in an infinite loop until the butto is released. Some controllers make a single blink at 255 in order to advise the user that maximum brightness is reached and he can immediately release the button. This is useful but optional. 

- Double press: normally not implemented in a standard dimmer because is unnecessary (normal dimmer has no other functions). Wled may leave this configurable i.e. to call a preset that random changes the color or reset the preferred white.

So the best way is to assign this way of working to the B1 with a zero setting on short and long pressure settings field. On the user interface the user should see a "*" going to a tip that explains that way of working.
This is my own honest opinion, but it follows the "standards" and I'm convinced that following the standards (if possible) means building the future of a project.

blazoncek, 2022-12-02:
FYI (caveat you have to release the button)
[image]

Mopfus, 2022-12-02:
> If you use button (with ID) 1 you get a ramping brightness using long press. 
So this function is now available?  The Idea is that it does like in this video: https://user-images.githubusercontent.com/80747454/160247481-01247e5e-1965-4df2-9e6a-96cf31842a19.mp4

> @evazzoler please provide a PR with a better (standardised) solution without hampering other functions or bloating the code.
I could show you my solution in ESPHome. But I guess that would not help, correct? Is there any possibility to help @Aircoookie with the coding? I know some python and I would really like to have the functionality that is shown in the video abvoe.

This was my solution under HomeAssistant running on a D1mini under ESPhome using "neopixel". This code does what you can see in the video. But I really don't know if this would work on WLED.

  # Hold Button
    - timing:
        - ON for at least 1s
      then:
        - homeassistant.service:
            service: input_boolean.turn_on
            data:
              entity_id: input_boolean.led_button_hold

        - if:
            condition:
              lambda: 'return (id(neopixel_light_esp).current_values.is_on());'
            then:
              - logger.log: "ESP is on!"
            else:
              - light.turn_on:
                  id: neopixel_light_esp
                  brightness: 15%
              - delay: 0.1s
        - while:
            condition:
              binary_sensor.is_on: d7_button
            then:
              - if:
                  condition:
                    lambda: 'return id(neopixel_light_esp).current_values.get_brightness() < 0.16;'
                  then:
                    - globals.set:
                        id: dim_state
                        value: '1'
                    - logger.log: "Dimstate = 1"
                    - delay: 0.1s
              - if:
                  condition:
                   lambda: 'return id(neopixel_light_esp).current_values.get_brightness() == 1;'
                  then:
                    - globals.set:
                        id: dim_state
                        value: '0'
                    - logger.log: "Dimstate = 0"
                    - delay: 0.1s
              - if:
                  condition:
                      lambda: 'return id(dim_state) == 1;'
                  then:
                    - light.dim_relative:
                        id: neopixel_light_esp
                        relative_brightness: 1%
                        transition_length: 0s
                    - delay: 0.02s
              - if:
                  condition:
                    and:
                      lambda: 'return id(dim_state) == 0;'
                  then:
                    - light.dim_relative:
                        id: neopixel_light_esp
                        relative_brightness: -1%
                        transition_length: 0s
                    - delay: 0.02s
              - if:
                  condition:
                    lambda: 'return id(neopixel_light_esp).current_values.get_brightness() < 0.16;'
                  then:
                    - light.turn_off:
                        id: neopixel_light_esp
                        transition_length: 0s
                    - delay: 0.1s
                    - light.turn_on:
                        id: neopixel_light_esp
                        brightness: 15%
                        transition_length: 0s
                    - delay: 1s                        
              - if:
                  condition:
                   lambda: 'return id(neopixel_light_esp).current_values.get_brightness() == 1;'
                  then:
                    - light.turn_off:
                        id: neopixel_light_esp
                        transition_length: 0s
                    - delay: 0.1s
                    - light.turn_on:
                        id: neopixel_light_esp
                        brightness: 100%
                        transition_length: 0s
                    - delay: 1s      
        - homeassistant.service:
            service: input_boolean.turn_off
            data:
              entity_id: input_boolean.led_button_hold

MrAnonymm, 2023-08-12:
Anything new?

MerlinBlatterer, 2023-09-08:
So did a single button Brightness controll feature get implemented or not?

I got this as a solution:
[image]

x 1: A=10
x 2: A=128
x 3: A=255
x Brightness: win&P1=241&P2=243&PL=~

Button Short goes to Preset increment
Button Long goes to Brightness

blazoncek, 2023-09-08:
> So did a single button Brightness controll feature get implemented or not?

Yes.
```c++
void longPressAction(uint8_t b)
{
  if (!macroLongPress[b]) {
    switch (b) {
      case 0: setRandomColor(col); colorUpdated(CALL_MODE_BUTTON); break;
      case 1: bri += 8; stateUpdated(CALL_MODE_BUTTON); buttonPressedTime[b] = millis(); break; // repeatable action
    }
  } else {
...
```

Knobiwahn, 2023-09-29:
hey can you explain to setup the dimming ... im newbee with wled
i control my led with an esp32 integrated touch function .... short touch = on/off, long touch = FX colour change
so i want to dimm with long touch

jnslmk, 2024-12-22:
The dimming feature is the default button action for button 1 to 3. If you configure your button in LED settings to be e.g. button 1, long pressing it will dim brightness as long as you haven't defined any other button action in the Time & Macros menu.

It is documented here: https://kno.wled.ge/features/macros/#buttons

`The default (built-in) actions for button 0 are short-press: toggle on/off and long-press: select random color. Long-pressing for more than 6 seconds will open the WLED-AP with the default password (wled1234).
For further buttons, the default action for short press is cycling effects, long press ramp brightness, and double press cycle palettes.`