wled/WLED #761: VU Meter Option IDEA

RAMBORG, 2020-03-07:
A vu meter part would be great,
Example VU-Meter from Cinelights (YouTube)

That's the last thing and Wled is perfect :)

mike2nl, 2020-03-08:
Hehe, not the last thing but a very good idea. Was missing that too.

Think about two groups and the second one works in reverse direction. That would be cool with audio support. But audio support must be come from somewhere else not on the esp8266. But a VU animation is one missisng part.

RAMBORG, 2020-03-08:
Microphone MAX4466 for example on Pin A0.

RAMBORG, 2020-03-08:
Cinelights use a Arduino mega, uno...... Without WiFi feature, no Webserver.

sjude68, 2020-03-10:
Ledfx on a Windows 10 computer within Anaconda or Ledfx on a raspberry pi works well for music lights.  I tried both.

Try this.
https://ahodges9.github.io/LedFx/

RAMBORG, 2020-03-10:
It's not the same -_- 
I mean a version with a microphone, without windows or a raspberry.

Best example is on YouTube cinelights vu meter.

Aircoookie, 2020-03-12:
Closing this in favor of #124 , which proposes the same feature. Will comment on that issue soon :)

closed 2020-03-12 by Aircoookie

DJprasenjit, 2022-10-21:
i managed to do it. wire less VU meter using WLED programme. and its working great . i will share video here. its so simple setup.
its only works on windows laptop or desktop pc. its shows the display color background or the sound effect vu meter. both are looks great.