wled/WLED #762: Change host name of WLED? Multiple devices with WLED on same LAN.

velkrosmaak, 2020-03-08:
As per the title - I can't find a way to change the hostname in the WLED UI. Is there a way? 

I've set a name for it on my dhcp server, but it still sends the name 'WLED' via MQTT, so I can't see my other WLED device which is on the same network as the names conflict. 

Any advice?
Thanks!

fishbone-git, 2020-03-08:
Mqtt settings are in sync interfaces (for client name and topic), wifi setup has the mdns name.

velkrosmaak, 2020-03-08:
The mdns only seems to work for Zeroconf, not the actual DNS name of the device. Thanks for the tip for sync interfaces!

closed 2020-03-08 by velkrosmaak

TheOnlyMatty, 2020-03-16:
In my experience changing "Server description" in the user interface settings changes the actual dns name.

BBaoVanC, 2025-03-14:
> In my experience changing "Server description" in the user interface settings changes the actual dns name.

Just found this from a Google search. It seems like now, you have to change the mDNS hostname option for it to also change DHCP DNS. With the mDNS hostname set to blank, it was using `esp32c3-9ccba8` as the hostname.

BBaoVanC, 2025-03-15:
Sorry, nevermind, enabling mDNS hostname was not having an effect. I am getting #4586 instead.