wled/WLED #770: Connection issues

bunnosaurusrex, 2020-03-13:
I have one nodemcu and 3 d1 minis flashed with both the 090b1 and 091 versions, they all flash and set up just fine and connect to my ssid, i can see them connected to my unifi AP and all of them have static IPs and i can control to them through home assistant but connecting to the wled ui page through any browser or the app is problematic at best and im not really sure how to proceed, any suggestions?

huggy-d1, 2020-03-16:
Be sure and use the latest build. There are some bugs in the older version plus since the latest release, some problems were found and fixed. One of those may relate to your specific issue. 

It would help to know more about your wifi setup. Is there a special character or space in the SSID?

triggerx, 2020-04-21:
Also having connection issues with 0.9.1, came here to look for suggestions/answers....

triggerx, 2020-04-22:
I have all but confirmed that this is issue is presented in a Unifi AP based network.  I'm 0 for 5 on NodeMCUs and 0 for 2 D1 Minis on my home Unifi based network when trying to run WLED 0.9.x.  Just keeps resetting.  They work fine on my RAVPower travel router AP.  However EVERY one of those devices work fine on my network when running WLED 0.8.x.  So, I'm almost ready to conclude that *something* in 0.9.x is just not compatible with *something* in my Unifi network.  Also, as an fyi, I have my APs set to 2.4GHz.

raoulteeuwen, 2020-04-26:
@Aircoookie just did some Googl-ing regarding Unifi and 8266. Found some references to explicit use "WiFi.mode(WIFI_STA)". Like

- https://github.com/esp8266/Arduino/issues/2371#issuecomment-311934307 (found through https://stackoverflow.com/questions/50301916/wifi-config-isnt-working-on-esp8266 )
- since i noticed some people reporting they had good results with Tasmota, i checked whether wifi_sta is used by Tasmota, and it is, https://github.com/arendst/Tasmota/blob/209c8f4eb73b92419f86f699754e9116fe7808b8/tasmota/support_wifi.ino 
- while searching i also found https://esp32.com/viewtopic.php?t=10291 stating "Calling "WiFi.disconnect(true)" immediately before attempting to connect (Wifi.begin()) makes the connection work correctly more consistently.". I noticed WiFi.disconnect(true) is also in the Tasmota code

You know best whether you maybe have already tried using wifi_sta and/or whether using code similar to Tasmota might be interesting to investigate/try.

triggerx, 2020-04-29:
Made a binary (thanks to LookAtTheShinyShiny in Discord) yesterday with "WiFi.disconnect(true);" added to line 322 in wled.cpp (just before WiFi.begin).... and that fixed my problem!  Ran WLED all night, and testing out other boards tonight that I've had problems with, and they're running perfectly!  Thanks @raoulteeuwen !

raoulteeuwen, 2020-04-29:
@triggerx what made you decide to pick that statement to add and try? Found https://www.arduino.cc/en/Reference/WiFiDisconnect explaining what it does, and wonder how that helps? Did you also experiment with WiFi.mode(WIFI_STA) ?

serenity182, 2020-04-30:
Maybe this related to the issue: 

WLED 0.9.1 on ESP8266 (amica)

Issue:
WLED is running in AP mode. 
User connects via: wifi settings -> landing page of WLED
User selects a effect and returns to the homescreen of the phone
After a random time the ESP reboots

If the wifi connection to WLED is deleted after setting the effect, no reboot occurs.

closed 2020-09-04 by stale[bot]