wled/WLED #771: Control segments via MQTT

velkrosmaak, 2020-03-13:
I would like to turn on and off, as well as set patterns for individual segments on my WLED device. Is that possible? Thanks!

Def3nder, 2020-03-16:
Hi @velkrosmaak, yes, you can controll segments via MQTT.

To do this you need to use the API functions of the HTTP API with sending a message to 
`[mqttDeviceTopic]/api`

The messages you need for this are documented here: [HTTP Request API](https://github.com/Aircoookie/WLED/wiki/HTTP-request-API).

You need to include either `&SS=` (or  `&SV=` in another request before) to select the segment and then add the things you want to control. 

BUT: if you main application (e.g. Home Assistant or NodeRED etc,) does support JSON, it's a lot easier using the [JSON API](https://github.com/Aircoookie/WLED/wiki/JSON-API) for dealing with segments.

velkrosmaak, 2020-03-19:
Hey @Def3nder thanks for the reply! 

I'm using Home Assistant to achieve this, turning on different parts of the segment depending on which people are at home. I'll look into doing it using the JSON API! If you have any example code to hand, i'd be grateful in the meantime. :) Thanks again!

viktor6, 2020-05-06:
Give an example

mushdawg, 2020-05-30:
Hello, I have 9 segments running the same effect 'CHASE'.   I have a plex integration which changes the effect when plex is being accessed.  Is there a way for all the segments to be triggered and changed ?

Aircoookie, 2020-05-31:
@mushdawg If all segments should run the same effect, you can select them all (tick checkbox next to the segment name) and the effect of all segments should change if you send an FX api call over MQTT.

mushdawg, 2020-05-31:
Im using home assistant, with plex integration, I dont think MQTT is used.
I have selected all the segments in the wled mobile app, however when the call service is triggered only segment 0 changes effect, the others remain the same.

Aircoookie, 2020-06-01:
In that case, it is probably interfacing not via MQTT, but with the integration. It should already support segments AFAIK, is your home assistant the latest version?

closed 2020-10-09 by stale[bot]