wled/WLED #773: 0.9. 1 - Remote behaviour

jonwilliams84, 2020-03-14:
Hi, awesome project!

I have upgraded a couple of my esp8266 devices to 0.9.1 today and the 24-key is now unresponsive. I have to press the action (e.g. RGB button) followed by the brightness up button for the Leds to change. This is the same for all buttons on the remote, including on/off. 

I have also tried a 44 key remote, and similar behaviour. 

Flashed back to WLED version 0.9.0-b1 and remote behaves again.

fishbone-git, 2020-03-14:
Please see #729.  If you are able to compile, you can use the latest code for a fix.

Aircoookie, 2020-03-15:
Hi, glad that you like WLED! Closing as duplicate. It is fixed in the latest source code, or wait until release 0.9.2 :)

closed 2020-03-15 by Aircoookie