wled/WLED #777: MQTT api, preset cycle on and off both send <cy>0</cy>

Skinah, 2020-03-16:
The Http api gives the correct state, however the state does not get published to the MQTT broker.

Cycle on should be 
<cy>1</cy>

instead it stays as

<cy>0</cy>

Def3nder, 2020-03-16:
Hi @Skinah, how did you check this ?

did you send an MQTT message to `[DeviceTopic]/v` ?
...and the returned XML did included <cy>0</cy> while the User Interface (in the browser) does show the Preset Cycle enabled (and it does cycle the presets) ?

Skinah, 2020-03-16:
I turn the light on and off with the webpage controls and the status mqtt message that gets sent by wled from the esp8266 does not reflect the state of the control.

If I use the http://ipaddress/win
It shows correctly how I am expecting it to work. The issue is the mqtt does not match what http displays and I only get <cy>0</cy> no matter the state of the control.

I do get the full status messages I have just shortened it to the important part which is <cy>0</cy> never changes via mqtt but it does via http.

Skinah, 2020-03-16:
Sorry misunderstood. Yes the controls function correctly and the presets cycle like they should. The only issue is the mqtt message does not match what the http api gives and I believe the mqtt is wrong. Because the message does not change it does not get sent which is why I have to use the power switch to get the mqtt message to send.

Aircoookie, 2020-03-18:
Hmm I believe that the issue could be that MQTT doesn't publish (so the values in MQTT are not updated) when the preset cycle is toggled, just when brightness or color is changed.
The same is likely the case in #776 for the <nl> attribute as well. Will look into it, MQTT should be updated if anything in the XML response changes.

swampdogmash, 2020-06-12:
Does sound like this will cause problems for MQTT users like myself. To get the status information into nodered for example for triggering other events. Openhab definitions also work better with MQTT. I'm having similar issues trying to get the MQTT and HTTP feedback loop to work writing my own "version" of what is not even a binding - just item files and rules.

Skinah, 2020-10-10:
Still an issue but not for me as I no longer use the mqtt api due to this issue. It effects multiple features of the API.

blazoncek, 2022-10-10:
Preset cycle has been removed in favour of playlists in 0.12 and above.

closed 2022-10-10 by blazoncek